silvanmelchior/RPi_Cam_Web_Interface #604: /dev/shm/mjpeg/user_annotate.txt

BETLOG, 2021-01-03:
At what systemctl target is "/dev/shm/mjpeg/user_annotate.txt" created?
I'm writing a service in python and after each reboot:
IOError: [Errno 2] No such file or directory: '/dev/shm/mjpeg/user_annotate.txt'

roberttidey, 2021-01-03:
The raspimjpeg process just looks for that file and if it is there it uses the contents to include in the annotation as required; it does not create it.

/dev/shm is a RAM based part of the filing system and will always start off empty at boot. It is used by raspimjpeg for temporary content that is being updated all the time to avoid wear on flash memory.

BETLOG, 2021-01-03:
Its being weird
Mind you, python is something i learn a little more of and then go back to bash for a few years, so maybe it's just me.
As someone more familiar with this than I am maybe you can see my oversight below?

If i ssh in, and execute the .py as a regualr user, or via sudo su it works fine. But if i reboot it fails:

I'm starting to think it may be easier to just figure out the bash i2c tools and rewrite the UPS_Lite.py (the bulk of the .py below) and just work from there.

```
pi@pizerocam2:~ $ sudo systemctl status piinfo.service
● piinfo.service - RPiWebCamInterface user annotation service
   Loaded: loaded (/lib/systemd/system/piinfo.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-01-03 22:19:05 AEST; 1min 25s ago
  Process: 272 ExecStart=/usr/bin/python /home/pi/piinfo.py (code=exited, status=1/FAILURE)
 Main PID: 272 (code=exited, status=1/FAILURE)

Jan 03 22:18:58 pizerocam2 systemd[1]: Started RPiWebCamInterface user annotation service.
Jan 03 22:19:05 pizerocam2 python[272]: Traceback (most recent call last):
Jan 03 22:19:05 pizerocam2 python[272]:   File "/home/pi/piinfo.py", line 64, in <module>
Jan 03 22:19:05 pizerocam2 python[272]:     with open(annotationPath, "w") as annotationfile:
Jan 03 22:19:05 pizerocam2 python[272]: IOError: [Errno 2] No such file or directory: '/dev/shm/mjpeg/user_annotate.txt'
Jan 03 22:19:05 pizerocam2 systemd[1]: piinfo.service: Main process exited, code=exited, status=1/FAILURE
Jan 03 22:19:05 pizerocam2 systemd[1]: piinfo.service: Failed with result 'exit-code'.
```

```
[Unit]
Description=RPiWebCamInterface user annotation service

[Service]
Type=simple
#Type=forking
ExecStart=/usr/bin/python /home/pi/piinfo.py
PIDFile=/tmp/piinfo.pid

[Install]
WantedBy=default.target
#WantedBy=local-fs.target

```

```
#!/usr/bin/env python
import struct
import smbus
import sys
import time
import RPi.GPIO as GPIO

import time
import datetime

from gpiozero import CPUTemperature
cpu = CPUTemperature()

import socket
hostname = socket.gethostname()

#from pathlib import Path
annotationPath = "/dev/shm/mjpeg/user_annotate.txt"
#Path(annotationPath).touch()

#import os
#os.system('touch annotationPath')

def readVoltage(bus):
        "This function returns as float the voltage from the Raspi UPS Hat via the provided SMBus object"
        address = 0x36
        read = bus.read_word_data(address, 0X02)
        swapped = struct.unpack("<H", struct.pack(">H", read))[0]
        voltage = swapped * 1.25 /1000/16
        return voltage

def readCapacity(bus):
        "This function returns as a float the remaining capacity of the battery connected to the Raspi UPS Hat via the provided SMBus object"
        address = 0x36
        read = bus.read_word_data(address, 0X04)
        swapped = struct.unpack("<H", struct.pack(">H", read))[0]
        capacity = swapped/256
        return capacity

def QuickStart(bus):
        address = 0x36
        bus.write_word_data(address, 0x06,0x4000)

def PowerOnReset(bus):
        address = 0x36
        bus.write_word_data(address, 0xfe,0x0054)
       
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(4,GPIO.IN)

bus = smbus.SMBus(1)  # 0 = /dev/i2c-0 (port I2C0), 1 = /dev/i2c-1 (port I2C1)

PowerOnReset(bus)
QuickStart(bus)

while True:
 st = datetime.datetime.fromtimestamp(time.time()).strftime('%Y-%m-%d--%H-%M-%S')
 with open(annotationPath, "w") as annotationfile:
  annotationfile.write(st + "  " + hostname + "%6.1fC" % cpu.temperature + "%4i%%" % readCapacity(bus))
#  annotationfile.write("%s  %3.1fC %4i%%" % (st,cpu.temperaturereadCapacity(bus))
 print "%s  %s  %3.1fC %4i%%" % (st,hostname,cpu.temperature,readCapacity(bus))
 time.sleep(1)
```

closed 2021-01-11 by BETLOG