silvanmelchior/RPi_Cam_Web_Interface #605: motion masks - do they auto refresh

BETLOG, 2021-01-09:
I have made a right click context menu to convert and upload motion masks.....
Assuming the filename is the same as that defined in the configs/menu; **do i need to** tell RPCWI to update the mask?
Reboot?
stop/start motiondetection? 
Or will it automatically use the new file without me doing anyhting?

roberttidey, 2021-01-09:
The motion mask file is loaded and activated whenever the camera set up is 'started'.

This happens when the overall software is first started and when particular commands are used which require fundamental changes to the set up.

These include the following commands px, qp, pv, bi, ru, md, rs, vp, cn. (See command summary on schedule page).

In summary just loading a new file or changing the file name will not trigger it to be used, but starting motion detection will cause it to be refreshed.

BETLOG, 2021-01-09:
I must be doing something wrong:
In light of what you said above it would be useful to toggle the md off and then back on again to refresh the mask image. Iv'e tried a few variants like these, but I'm getting no output from FIFO, and no reaction from the browser interface button (after i refresh it)
What's wrong with my syntax/permissions?

```
printf "%s\n" "md 0" >/var/www/html/FIFO1
printf "%s\n" "md 1" >/var/www/html/FIFO1

printf "%s\n" "md=0" >/var/www/html/FIFO1
printf "%s\n" "md=1" >/var/www/html/FIFO1

echo "md 0" >/var/www/html/FIFO1
echo "md 1" >/var/www/html/FIFO1

echo "md=0" >/var/www/html/FIFO1
echo "md=1" >/var/www/html/FIFO1

```

I'll be doing this via ssh as part of the mask upload script.

roberttidey, 2021-01-09:
The main command pipe is FIFO. One can also use FIFO11 -FIFO19 if required to avoid multiple applications trying to write to the same pipe.

FIFO1 is not a command pipe it is used to provide communication between the raspimjpeg and the scheduler to pass motion detection events. 

See wiki under pipe and motion detect for details.

Commands being processed correctly will put stuff into the scheduler log file.

Note that I will may add the camera restart sequence to the mf command so that a change in the mask filename would trigger a reload. However, this would not effect uploading a new mask file with the same name. That would still need something like the motion detect off / on sequence.

BETLOG, 2021-01-09:
> FIFO1 is not a command pipe

Oh....I guess this threw me:
[image]
...It still does.
But blind obedience works, so thats good enough for me.
```
ssh pizerocam3 "echo md=0  >/var/www/html/FIFO;echo md=1  >/var/www/html/FIFO"
```

FYI: 
I don't think that relying on filename changes to trigger mask updates will always be useful.
The GUI 'OK' button press would seem to be best suited to triggering an update.

closed 2021-01-11 by BETLOG