silvanmelchior/RPi_Cam_Web_Interface #606: record pre-motiondetect trigger frames

BETLOG, 2021-01-11:
Would it be possible to include N frames  or N seconds of video in the recording **before** motion detection was triggered?

I either get far too many false positive triggers because of low threshold settings, or videos start with someone in the centre of frame.
I'm using a very restrictive mask, and a low threshold so I expect people entering the edges of the frame to register before they have spent several seconds walking into the middle of the camera.

roberttidey, 2021-01-11:
That's the purpose of the buffer setting. If you set it to about 2 seconds (2000) then you will get the event leading up to the motion trigger.

It is not, however, a precise amount of pre-trigger frames due to it being based on maintaining a continuous circular buffer of video recording which is pre-pended onto the actual recording. As this is compressed data then it represents a  variable amount of recording time and the buffer size is based on an estimate of this. You may want to play around with the size of the buffer time to get the amount of pre-trigger that suits your needs.

BETLOG, 2021-01-11:
Excellent, thanks.
I was looking at the 'motion settings' tab, no wonder i didn't notice that.

closed 2021-01-11 by BETLOG