silvanmelchior/RPi_Cam_Web_Interface #607: watchdog isnt working

username7291, 2021-01-13:
I see right now a image from 2 hours ago. Its crashed cause of low disk space (23MB log is created with cant write h264). The watchdog should reset the Camera/System if the preview image is always the same or? Even if i delete all files the preview stays the same.

roberttidey, 2021-01-13:
The watchdog triggers based on the preview image file in /dev/shm/mjpeg not being updated for longer than the watchdog period. This can happen if the camera subsystem halts for some reason.

If you have run out of disk space then that can cause all sorts of other errors and some of these may prevent the watchdog from working.

Deleting files AND restarting the system should restore operation.

I would recommend using scheduler purge rules to manage the disk space to ensure that you don't encounter that sort of error.  If you don't want it done automatically then you could set up a maintenance script to check and warn you if space is getting low.

username7291, 2021-01-13:
I understand the watchdog reset too on X Errors (like on settings). There was really alot Errors in the log.
Cause of issues with the fps(max.22 with Motion detection enabled),sometimes 3-4 identical frames in the video, and not harm the SD Card i use /tmp/ in Memory for h.264 and SD only for MP4. tmp was full, not the SD Card. Im not sure what happend(there should be no movement maybe Wind). But it streams until limit, split the file, streams again. 
Maybe the task that streams from /dev/shm/mjpeg to the web interface was frozen?
It puts out 21 times/second "Error: Could not write all bytes h264", maybe it has no time todo anything other. There was 3 Big files in /tmp and about 8 0Bytes files. Seems MP4Box stops processing, maybe it need the amount of Memory of the filesize and dont stream like common programs(150MB available after tmp is 100% full, 200MB file) and the h264 source files never gets removed on failure. 

I wrote a own watchdog, that checks the filetime of /dev/shm/mjpeg, the free MB of /tmp and the gpu memory(rasbian bug 1 time out of gpu memory, omx never works again until reset).
If one fail it reboots.

roberttidey, 2021-01-13:
As /tmp is RAM based it is using up the real CPU memory of the Raspberry which can be much more limiting than the SD card memory. I have not tried to use it for the direct h264 data for that reason.

If you are in danger of getting too long motion triggered video then you should try tuning the motion parameters. Temprarily including %f and %c in the annotation string can help with this as it allows you to see the factors triggering motion detection in real time. %c is the amount of motion change detected and %f is the count of contiguous changed frames being seen.

username7291, 2021-01-15:
I set the split time shorter for the case if mp4box has an huge memory usage. Usual MP4Box is faster than realtime, takes about 23sec for 100sec video.
I exchange the original 20cm HQ Camera cable against a shorter one, but still have errors in the Video.
Is this as expected with internal motion detection running or is this HQ Camera related?
For example only 22fps, ghost images from an moving object on a position it was earlier or double frames?