silvanmelchior/RPi_Cam_Web_Interface #610: Running FIFO commands and sleep from userbutton, FIFO

ffcruz85, 2021-02-10:
So, pretty sure this is something related to issue #477 - but quite honestly, I couldn't understand the resolution there, and I am also  wondering if it might be related to how the macro file was created (I just did sudo nano macroname.sh without doing anything like chmod after)

so, basically, I wanted a few userbuttons to start recording, wait a few minutes, stop recording (because I always forget). so I created this userbutton file:
`Record_2_Minutes,2minvideo.sh,btn btn-primary,style="width:10%" autofocus`

and this is the macro file, and yes, it's in the macros folder:
```
#!/bin/bash
echo 'ca 1' >/var/www/FIFO
sleep 1m 55s
echo 'ca 0' >/var/www/FIFO
```

I also tried leaving only the first 'ca 1' line for testing purposes in case sleep was the issue, also didn't work.

The result:
Clicking the button registers the following in the log, but nothing else happens:
'{2021/02/10 15:45:54} Executing macro /var/www/macros/2minvideo.sh &'

To confirm it was not an UI button issue, I watched the browser console while clicking and also tried running the macro from the terminal over ssh with the following:
`echo 'sy 2minvideo.sh' >/var/www/FIFO`

same result, shows in log, nothing happens
PS.: By log, I mean scheduleLog.txt - is there another one?

Thanks again @roberttidey for all the awesome work over the years. I wish I knew some PHP to offer help :)

roberttidey, 2021-02-11:
It is important that the execution privileges and ownership of macro script files are set after they are created. They will not work until this done.

Macro scripts are executed by the web server which does this under the user www-data who has limited rights as this ensures safety against attacks that may try to get scripts to do malicious activities. If one needs the script to have particular privileges to perform specific extra actions then those can be enabled and can be made specific to that particular macro script. 

In your case the echo actions are OK and don't need extra privileges, however the script file still needs to have the right ownership and overall execution privileges.

As described chown sets the ownership and chmod sets the execution privileges.

scheduleLog.txt is used as the main log

ffcruz85, 2021-02-15:
Ok, does that work - sorry for the basic question, did not want to break my whole setup. CLosing the issue, but a follow up question if I may: what's your preferred way of backing up configuration for an installation?

closed 2021-02-15 by ffcruz85