silvanmelchior/RPi_Cam_Web_Interface #611: Get current configuration value from shell script macro

ffcruz85, 2021-02-15:
I am trying to get an existing configuration setting to be used in a macro. Here's the scenario: my camera is running a timelapse at X seconds interval, I want to hit a userbutton to temporarily run it at a different interval, then set it back to whatever it was before. Current code:

```
#!/bin/bash
# Get current value
TLinterval=XXXXX
# turn off current timelapse
echo 'tl 0' > /var/www/FIFO
# timelapse is in tenths of a second, setting to 3 sconds
echo 'tv 30' > /var/www/FIFO  
echo 'tl 1' > /var/www/FIFO
sleep 45m
echo 'tl 0' > /var/www/FIFO
#return timelapse to user's previously saved interval
echo 'tv $TLinterval' > /var/www/FIFO  
echo 'tl 1' > /var/www/FIFO
```

thanks!

roberttidey, 2021-02-15:
The basic default configuration parameters are kept in /etc/raspimjpeg

When you make a change to setting then it is stored in the uconfig file which is in the web folder (e.g. /var/www/html/uconfig)

This is just a text file with one parameter stored per line. The time lapse interval will be a line like
tl_interval 20

So your macro script could look through the uconfig file and extract that value. The awk utility can simplify this

Note that if you change a setting back to the default value in /etc/raspimjpeg then the line will be removed from the uconfig file. I.e the absence of a parameter line means that it is at the default value.

You can back up any set up by just saving the uconfig file, but if you have changed any of the defaut values in /etc/raspimjpeg then you should back that up too.

ffcruz85, 2021-02-15:
Great, I will try that out. I am trying to do some more advanced macros as well as potentially reading/managing some configurations via my home assistant. Thanks again for the help

closed 2021-02-15 by ffcruz85