silvanmelchior/RPi_Cam_Web_Interface #619: Can I reinstall remotely?

Woody565, 2021-04-29:
Hi,

I have Rpi Cam installed on a Pi4 insode a bird box, there is currently a bird  sitting on eggs in the box so I cannot access the SD card to format it and start again.

If I try to adjust any camera settings I get an error "Error in RaspiMJPEG:" 

Can I connect with putty and run the install command again to install a new copy of the software?

I also believe this could be a camera connection issue but there is nothing I can do about that at the moment.

Thanks for any help

roberttidey, 2021-04-29:
I do all my installs remotely. If you can access it via putty then you can certainly re-install or perform any other software repair needed.

Woody565, 2021-04-29:
thanks for the help, I ran the follwing commands but cant get a new install

git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git - it says 
fatal: destination path 'RPi_Cam_Web_Interface' already exists and is not an emp                                                                                        ty directory.

then I ran 
cd RPi_Cam_Web_Interface

then
./install.sh - which returns 
Reading package lists... Done                          
Building dependency tree                               
Reading state information... Done               
apache2 is already the newest version (2.4.38-3+deb10u4)u
gpac is already the newest version (0.5.2-426-gc5ad4e4+dfsg5-5)  
gstreamer1.0-tools is already the newest version (1.14.4-1)
libapache2-mod-php7.3 is already the newest version (7.3.27-1~deb10u1)
motion is already the newest version (4.1.1-1.1+b2)
php7.3 is already the newest version (7.3.27-1~deb10u1)
php7.3-cli is already the newest version (7.3.27-1~deb10u1)
zip is already the newest version (3.0-11)
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded
Your custom raspimjpg backed up at /etc/raspimjpeg.bak

I dont think it has reinstalled?

Am I correct?

roberttidey, 2021-04-29:
It sounds like it just installed the version that was already present. The 0 upgraded messages just refer to support packages like Apache php etc which won't change.

git clone will not overwrite the install folder.

To install the latest version I suggest you completely remove the RPi_Cam_Web_Interface folder (I think rm -rf RPi_Cam_Web_Interface)   from your home folder. Then do the git clone and run the install.sh  You will still see 0 upgrade messages for things like Apache but that is normal.