silvanmelchior/RPi_Cam_Web_Interface #620: Support for split screen

MohammedRehabS, 2021-05-01:
We need a split screen kind of video feed of the same camera on the webpage, how is it possible to do that?

roberttidey, 2021-05-01:
You will have to provide some more detail on what you are trying to achieve. Split screen is too ambiguous.

MohammedRehabS, 2021-05-01:
We need something which could give a VR like view . So we need two streams of the same camera side by side. Something like a split screen mode.

roberttidey, 2021-05-02:
If you want two stream windows showing identical content side by side then you should be able to achieve that by editing the index.php page and creating a second preview window (see around line 299 img mjpeg_dest). This would need a separate id. Then in the javascript script.js then the updatePreview function would need editing to also update this second img object.

MohammedRehabS, 2021-05-03:
Worked out perfectly , thank you very much. 

Can you please tell us how can we achieve multiuser login as well.

roberttidey, 2021-05-03:
In the wiki see section 4.11 and 4.15

closed 2021-05-03 by roberttidey

MohammedRehabS, 2021-05-03:
Thank you very much