silvanmelchior/RPi_Cam_Web_Interface #624: Problem with Filter dropdown on preview.php page

wn-dev, 2021-06-24:
The Filter dropdown on the preview.php page doesn't allow selecting files that are between 144 and 192 hours old.  There's an option for "120-144 hours old" and an option for "192+ hours old", but nothing in-between.  See the red-circled section in the below screenshot.  Maybe the option for "192+ hours old" can be changed to "144+ hours old" to fix this problem?

[image]

roberttidey, 2021-06-24:
Thanks for pointing this out. This list labels are calculated on the fly and there was a coding error for the last entry. It was actually 144+ despite its label.

Corrected in latest version

wn-dev, 2021-06-24:
@roberttidey Thanks for fixing the label problem.  Using the latest code, selecting the "144+ hours old" filter results in files aged 168+ hours being shown in the results.  Files aged 144 to 168 hours are missing in the results, which suggests that the filtering logic for this case is off by 24 hours.

roberttidey, 2021-06-25:
OK. Checking into that further.

roberttidey, 2021-06-25:
I think latest version now corrects this.

wn-dev, 2021-06-26:
Yes, the latest version fixes the problem.  Thanks @roberttidey for the quick resolution.

closed 2021-06-26 by wn-dev