silvanmelchior/RPi_Cam_Web_Interface #625: Unable to change motion settings

tjwpfr, 2021-06-28:
Everything seems to be running fine, I get no errors, everything functions as expected except when I change the motion detection settings and hit save it just resets everything back to the default values. 

If it's an issue with the web gui is there a config file for motion I could manually edit?

I'm using  a fresh up-to-date raspberry pi OS install on a pi 4 B using the pi cam high quality.

roberttidey, 2021-06-29:
It sounds like you are using the external motion detect method.

If so then I strongly recommend switching to the internal motion detection which works much better and also has a much lower load on the cpu. (Change motion detect mode under camera settings)

The external motion detection method relies on the motion process and uses a web api to change settings. It is possible that this api has changed which is what would cause the problem with saving. I will look into that but it will be low priority as the internal method is much better.

tjwpfr, 2021-06-29:
You are correct, problem is limited to the external motion detection. I would recommend making the internal one the default, it works great.

closed 2021-06-29 by tjwpfr