silvanmelchior/RPi_Cam_Web_Interface #626: Recording RAW image using Web Interface

mirzat-aston, 2021-07-16:
Hi folks,

I need help with recording a RAW image instead of JPG format. RAW can be in YUV or RBG format. Kindly guide me on how to use this web interface for recording RAW images.

I would really appreciate your help as I am beginner and trying to learn this platform.

Regards,
Taimur

roberttidey, 2021-07-16:
There is already a raw layer option under camera settings. When this is on it appends the raw camera data onto the end of the jpeg. The jpeg is still viewable and the raw data may be extracted from the file by various techniques.

See for example https://www.raspberrypi.org/blog/processing-raw-image-files-from-a-raspberry-pi-high-quality-camera/

Note that you could automate the extraction of the raw data by writing a macro command which is executed after each image is taken.

dhruvilkpatel, 2022-03-03:
import time
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (100, 100)
    camera.start_preview()
    time.sleep(2)
    camera.capture('image.data', 'yuv')