silvanmelchior/RPi_Cam_Web_Interface #627: Restart raspiMJPEG popup after a few hours

GavChap, 2021-07-29:
[image]

This happens after about 1-2 hours of running.

It's a third party pi cam module set to 1296 x 972 (max the camera can handle) otherwise it's all default settings.

There's a bunch of this in the syslog:

```Jul 29 17:50:02 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Jul 29 17:50:02 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Content-length present
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Content-length present
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Jul 29 17:50:03 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Jul 29 17:50:04 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Jul 29 17:50:04 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Content-length present
Jul 29 17:50:04 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Jul 29 17:50:04 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Jul 29 17:50:04 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
```

GavChap, 2021-07-29:
Forgot to add this is a Raspberry Pi 4, 2Gb model.

roberttidey, 2021-07-29:
This sort of error is most often caused by either power problems or an unreliable connection between the raspberry pi and the camera module.

Check you are getting a good solid 5V at the Pi. The Pi4 takes quite a bit of power and the camera module itself takes a bit more. With a Pi4 the most reliable method is the official Raspberry Pi4 power supply. If you are using your own power supply check it can supply sufficient current and in particular ensure the USB cable is good and short. Many USB cables can have significant resistance and cause voltage drop at the Pi.

For the camera module check the ribbon connections at the both the Pi and camera ends and also check and press down the tiny ribbon cable on the module itself.

If neither helps do you have access to another camera module to check it is not a faulty module causing the issue.

I have several camera which have been running for very long periods without issue so this type of error is unlikely to be a software problem.

GavChap, 2021-07-30:
I'm using the official Pi 4 power supply, so I doubt it's that, I'll swap out the ribbon cable, and if I still get the problem I'll have to buy a new module, thanks.