silvanmelchior/RPi_Cam_Web_Interface #628: cam subfolder broken with nginx

druck13, 2021-08-05:
With versions 6.6.14 onwards selecting a _cam subfolder_ in the installer no longer works, as the url http://host/subfolder is not found (404). This is because the sub folder is now appended to the root field in /etc/nginx/sites-available/rpicam e.g. `root /var/www/subfolder;` On version 6.6.13 and prior the root was set to `root /var/www;` and the url http://host/subfolder works.

A work around is to edit change the root field in /etc/nginx/sites-available/rpicam to `root /var/www;` and do a `systemctl restart nginx.service`

roberttidey, 2021-08-05:
This is due to a pull request entered for issue 600 where the webport was not being honoured. The request code change also set the root as you saw.

Some users prefer to avoid the subfolder in the url which is why this change was probably included but it is inconsistent with the other web server set ups so I propose to revert that part.

It is line 226 in install.sh

druck13, 2021-08-06:
Version 6.6.16 upgraded without problems across my 6 pi's - thanks!

closed 2021-08-06 by druck13