silvanmelchior/RPi_Cam_Web_Interface #629: The preview.php page's Preview and Thumb size inputs behave strangely

wn-dev, 2021-08-16:
When a value less than 100 or greater than 1920 is entered in the Preview input box and the Update button is clicked, the value gets reset to 640.  Would it make more sense to reset the value to 100 if a value less than 100 is entered, and to reset the value to 1920 if a value greater than 1920 is entered?

The Thumb input box behaves in pretty much the same unexpected way.  When a value less than 32 or greater than 320 is entered and the Update button is clicked, the value is reset to 96.  Would it make more sense to reset the value to 32 if a value less than 32 was entered, and to reset the value to 320 if a value greater than 320 was entered?

In addition, isn't the default thumbnail size 512 in width?  I'm assuming it's the "Width" value from the Camera Settings section of the index.php page.  If 512 is indeed the default thumbnail width, would it be better for the Thumb size input on the preview.php page to use 512 instead of 320 as the upper limit?

roberttidey, 2021-08-18:
Thanks for your comments.

The range checks for preview and thumb size were setting back to default if range was exceeded either way.

I have improved that so it sets to minimum or max value as appropriate.

I have not changed max thumb size as it performs a different purpose on the main screen and on download screen where there can be many thumbnail icons.

wn-dev, 2021-08-19:
@roberttidey Thanks for updating the code.  I've tried it out and it now works as expected.

closed 2021-08-19 by wn-dev