silvanmelchior/RPi_Cam_Web_Interface #630: How can one build raspimjpeg?

2010019970909, 2021-08-17:
Hi,

I'd like to know how is build raspimjpeg, but I don't really know where to start? Could someone guide me through?
Also, it there a documentation for raspimjpeg?

Thanks in advance,
Vincent

roberttidey, 2021-08-17:
raspimjpeg is hosted at https://github.com/roberttidey/userland which is a fork of raspberry userland with the raspimjpeg source files under host_applications/linux/apps/raspicam   (raspiM*)

To build it git clone this userland onto a raspberry pi which should give you a userland folder under your user home. 

You need to have cmake installed. Install that with apt-get if not already there. The gcc compilers should already be there.

After that just cd into the userland folder and run ./buildme

This will build the whole of userland including raspimjpeg. The binary file will be found in the userland bin folder and if you have turned off the camera system then it will also update the operational version under /opt/vc/bin/

The first build will take a little while but subsequent builds will be much faster.

2010019970909, 2021-08-19:
Thanks for your answer. I'll try to build it on a Pi Zero (the RPi 4 I had just died so I have to replace it...). I will probably try to build it with a Github action (on an ARM runner) later.

I have a question about userland. Why not using the raspberry pi userland and solely keep the raspicam folder?

I'll try the following on a fresh install (Raspberry OS Lite) :
```bash
sudo apt-get upade && sudo apt-get upgrade
sudo apt-get install -y git cmake
# Enable camera after next reboot (optional)
# sudo raspi-config nonint do_camera 0 && sudo reboot
git clone https://github.com/roberttidey/userland.git
cd userland
sudo chmod +x ./buildme
./buildme
cd bin
```
I have two last questions (I'm more developing Python apps, so I'm not used to this kind of builds). First, how much time can it take. Second, what is the operational version?

Thanks in advance,
Vincent

roberttidey, 2021-08-19:
I use the whole userland folder just to make it easier to do the build as it has all source files for things like the MMAL camera interface and ensures all external references are resolved. In principle it would be possible to strip it down but I haven't done that.

It will take a while to build the first time on a Pi Zero. I normally build on a Pi3 and this takes maybe 5 - 10 minutes for a first build but subsequent ones are around 30 seconds or less as only changed files are processed. A Pi4 would be even faster. I guess a Pi Zero could be maybe 2 or 3 times slower.

The version of raspimjpeg is kept in the RaspiMJPEG.h and is currently 5.9.5. This gets logged into the scheduler log on start up.

2010019970909, 2021-08-21:
I was able to build it on a RPi 3 which was lying around (part of another project…). And on the Pi Zero and indeed, it took like less 1 hour to build on the Pi Zero and around 10 minutes on the RPi 3. I'll try to compile it with qemu ([here](https://github.com/eLAB-Birdhouse/birdhouse_raspimjpeg/actions)) but for the moment cmake is throwing errors. I may have to [recompile/compile](https://gitlab.kitware.com/cmake/cmake/-/issues/20568) cmake before.

2010019970909, 2021-08-21:
I forgot to ask. Is there a documentation for the usage of raspimjpeg? Or everything is on Raspberry Pi forums (I don't know where to look at) ?

roberttidey, 2021-08-21:
There is no specific documentation for the raspimjpeg process.

The following may help.

It basically consists of the following source files.

RaspiMJPEG.h - main header file containing all the external calls and variables
RaspiMJPEG.c - main shell setting up and controlling the command pipe and supporting the configuration
RaspiMCam.c interface to all the camera functions via the MMAL interface. Sets up video and still image processing and all camera configuration calls via MMAL. Part of the power of raspimjpeg relies on the fact that one can have multiple streams being processed by MMAL and these rely on the power of the GPU and cause little burden on the CPU even on a Pi Zero. So for example there can be streams of data sent for video recording, motion detection, thumbnail generation, still image recording and motion detection.
RaspiMCmds.c - Processes all the command pipe functions
RaspiMUtils.c - set of utility functions. E.g. logging, status handling, naming and script support
RaspiMMotion.c - internal motion detection support via vector movement processing

The external interface to all the functions is via a command pipe (one can set up several parallel pipes to give multiple access from different applications). The pipe takes 2 character command keywords and associated parameters. These are documented in the wiki and a summary is accessible on the schedule page.

If you have any specific questions I will try to answer them.

2010019970909, 2021-08-23:
Thanks, I was more looking for how to set and use the pipe, how to set the configuration, etc.

I've seen the readmes. [This one](https://github.com/2010019970909/userland/blob/master/host_applications/linux/apps/raspicam/README_RaspiMJPEG.md) is the more pertinent, but it was before the switch to a config file ? OK, I see I've missed some part of the [Wiki](https://elinux.org/RPi-Cam-Web-Interface#Scheduler)…

For the rest, I've set up a github action on my fork to lint (with super-lint) the code you pointed to me (and the readmes different from the readme.md). I'm far from being good at programming in c/c++, but the linter pointed me to some error and I also tried to address some compiler warnings.

The main issue is a missing header guard in `RaspiMJPEG.h` (I just added `#pragma once`). I've also seen a variable named `string` (I think it is locally defined, but it could maybe be an issue later -- at least the linter don't like that).

There are also some functions that are not recommanded according to cpplint (the linter used by super linter for c/c++). For example:
It promotes the use of `localtime_r()` instead of `localtime()` (because the latter is not thread safe).
It also prefers `strtok_r()` to `strtok()` for the same reason (see this [question](https://stackoverflow.com/questions/22210546/whats-the-difference-between-strtok-and-strtok-r-in-c) on SO).
Finally, it also recommand `snprintf()` over a few other functions.

There are some missing includes. Among which the ones for the `wait()` function inside the definition of the `monitor()` function triggers a warning.

Since I'm using Visual Studio Code (which has a lot of functionalities, like running on the remote Pi while being on a decent computer with a GUI to edit your code directly on the Pi), I formatted the code using [modified settings](https://stackoverflow.com/a/50489812/10949679) (`"C_Cpp.clang_format_fallbackStyle": "{ BasedOnStyle: Google, IndentWidth: 4, ColumnLimit: 0}"`). It allows to simplify the correction of cpplint warnings and errors (note that you can also add cpplint as an extension to Visual Studio Code).

If you wish, I can do a PR with my changes, but they are not tested correctly. It could also be interesting to build the binary within a github action to ensure repeatability and transparency (not that I don't trust you).