silvanmelchior/RPi_Cam_Web_Interface #631: Simple scheduler to save images every N seconds

guidocioni, 2021-08-30:
I want to do something really simple but somehow cannot find a way to make it work :)
My idea is to save a capture every N seconds (for now it's 3600, so 1 hour) in the media folder. 

I tried this scheduler 

[image]

which worked the first time
```
[2021/08/29 12:45:46] Scheduler loop is started
[2021/08/29 12:45:47] New period detected 0
[2021/08/29 12:45:47] Send im
{2021/08/29 12:45:47} Capturing image
[2021/08/29 12:45:49] Scheduled management tasks. Next at 1630241147
```
 but then stopped recording images. 
```
[2021/08/29 15:45:51] Scheduled management tasks. Next at 1630251950
[2021/08/29 16:45:51] Scheduled management tasks. Next at 1630255551
[2021/08/29 17:45:52] Scheduled management tasks. Next at 1630259152
[2021/08/29 18:45:53] Scheduled management tasks. Next at 1630262753
[2021/08/29 19:45:54] Scheduled management tasks. Next at 1630266354
[2021/08/29 20:45:55] Scheduled management tasks. Next at 1630269955
[2021/08/29 21:45:56] Scheduled management tasks. Next at 1630273556
```

What am I doing wrong?

roberttidey, 2021-08-30:
You put an im command in the period start field. This command is executed whenever that period starts. As the period is set to AllDay that only occurs once when the scheduler starts up or when you change settings.

If you want an image capture every N seconds then there are several ways to achieve that. Here are three.

1) Use time lapse instead of scheduler. This will take an image every time lapse interval. So if you set interval to 3600 and start time lapse then it will take an image once per hour. Separate images will be taken into the media folder but they will all be grouped under the one time lapse thumbnail.

2) Use management interval in the scheduler to run a macro script at regular intervals to perform the capture. 
A suitable script can be made as follows from a terminal window
a) navigate to macros folder [ cd /var/www/html/macros ]
b) copy the test.sh macro [sudo cp test.sh capture.sh ]
c) edit test.sh to send im command into the command input pipe [ sudo nano capture.sh ]
   change the second line to be echo 'im' > /var/www/html/FIFO
   save file [CTL_O, CTL-X ]
   Note that if you have used a different subfolder than html then change the echo command to suit
d) make script owned by www-data [sudo chown www-data:www-data capture.sh
e) make script executable [sudo chmod 744 capture.sh ]
Now set management interval to what you want (3600) and management command to capture.sh
Save settings
Script will now run once per hour and perform a capture. As they are separate captures rather than a timelapse they will each have individual thumbnails.

3) Use some external process to perform the echo 'im' command whenever you want a capture. E.g crontab can perform this command at whatever intervals you want.

guidocioni, 2021-08-30:
Dear @roberttidey,
thanks a lot for the quick reply.
As I only installed this yesterday I still have to get use to it, so apologize if I'm using some functions not the way that they're supposed to :) 

> 1. Use time lapse instead of scheduler. This will take an image every time lapse interval. So if you set interval to 3600 and start time lapse then it will take an image once per hour. Separate [...]

True, but I wanted something that can be controlled with a crontab kind of thing and not that requires to be manually activated in the web interface. As far as I understand the only way to make a timelaps is to press on "start timelapse" in the web interface and this will be reset in case of restart...right? 

> 2. Use management interval in the scheduler to run a macro script at regular intervals to perform the capture.
>    A suitable script can be made as follows from a terminal window
>    a) navigate to [...]

Thanks, I was missing a clear example like this. I think that's what I'm going to do. 

> 3. Use some external process to perform the echo 'im' command whenever you want a capture. E.g crontab can perform this command at whatever intervals you want.

This is basically the same as in 2. only that here the scheduling is managed by crontab but one still has to do echo 'im' > /var/www/html/FIFO, right?

Regardless of the method is there any way to capture in the bash script the output filename of the picture so that I can pass it to other operations? I would like to avoid having to list the images in the /media folder and choose the last one (or the most recent one): it's in your examples but and it will work most of the times but it's not 100% robust.

roberttidey, 2021-08-30:
For the time lapse method. If you put a tl 1 in the allday period start then that will be executed when the scheduler starts up and kick off the time lase operation without performing a manual operation.

Yes. The crontab comment about the echo is correct. Pretty much everything is controlled by the command pipe so can remote control it from external programs. See the pipe command reference either in the wiki or a summary in the scheduler window.

For the file name part one method is to utilise the job macros that can get run when particular operations are done including the start and end of image capture. So end_img.sh is run  (if it exists in macros folder) after the capture of every image. These macros are passed the capture filename involved. See the Job Macros section in wiki for more details.