silvanmelchior/RPi_Cam_Web_Interface #633: Have problem with start_vid.sh

markxyx, 2021-09-08:
I add a line to start_vid.sh. I ran it in terminal with no problem. It created a audio.wav file in /var/www/media as what I asked it to do.

[image]

When I use web interface to record a video. Nothing happened. I checked the log. It shows Raspimjpeg executed start_vid.sh. what did I missed?

[image]

roberttidey, 2021-09-08:
For a macro script to execute successfully under the web interface it must satisfy three conditions.

1. It must be executable. This is achieved using the chmod command. You must have already done this as you are executing it from a command line.
2. It must be owned by user www-data. The web server runs as user www-data and anything executed by it must have the same owner. This is for security reasons as user www-data has limited privileges to stop nasty things being done. You make script owned by www-data by using the chown command e.g.
chown www-data:www-data /var/www/macros/start_vid.sh

3. Any processes actions within the script must be allowed for user www-data otherwise they will not successfully run. So user www-data may need to be explicitly given permission for run arecord. One way to achieve that is to add the process to the sudoers file /etc/sudoers.d/RPI_Cam_Web_Interface  This already has entries to allow the web interface to run the shutdown and date commands. Use those as an example.

markxyx, 2021-09-09:
You are legend. It works.

closed 2021-09-09 by markxyx