silvanmelchior/RPi_Cam_Web_Interface #634: Make PRI Cam Web Interface

markxyx, 2021-09-12:


markxyx, 2021-09-12:
Follow these steps will add the ability of recording video with audio to RPI Cam Web Interface have. 
The concept is calling alsa to start recording audio when you hit "start video record" in web interface. Than convert it to boxable MP3 file. This will take sometime for orange "stop recording" buttom to turn blue. 
******Only hit the orange stop recording buttom once and wait until convertion and boxing finish******
This works for me. Hope this will work for you too. Good luck.

## Install & update alsa MP4Box and lame
# MP4Box is pre-installed by PRI Cam Web Interface
sudo apt-get install update gpac
sudo apt-get install update lame
sudo apt-get install update alsa

## Setup audio recording
# Creat and edit alsa system-wide config file /etc/asound.conf
sudo nano /etc/asound.conf
# follow setting should fit most usb audio devices that with input or/and output
 pcm.!default {
   type asym
    playback.pcm {
      type plug
      slave.pcm "hw:1,0"
    }
    capture.pcm {
      type plug
      slave.pcm "hw:1,0"
    }
 }
# If this dose not work. Google how to add a .asoundrc file for user www-data. 

## Add user www-data to audio group
sudo usermod -a -G audio www-data

## Creat start_vid.sh
sudo nano /var/www/macros/start_vid.sh

## Copy & paste follow lines to start_vid.sh
#!/bin/bash
#Call arecord to start recording audio and creat an audio.wav file
#modify parameter to suit your hardware
arecord -f S16_LE -r 48000 -c 2 /var/www/media/audio.wav

## Change owner & permission for start_vid.sh
sudo chown www-data:www-data /var/www/macros/start_vid.sh
sudo chmod 764 /var/www/macros/start_vid.sh

## Creat end_vid.sh
sudo nano /var/www/macros/end_vid.sh
## Copy & paste follow code to end_vid.sh
#!/bin/bash
# Stop recording with pkill
pkill arecord
# Convert audio.wav to audio.mp3 for boxing
lame -V2 /var/www/media/audio.wav /var/www/media/audio.mp3

## Change owner & permission for end_vid.sh
sudo chown www-data:www-data /var/www/macros/end_vid.sh
sudo chmod 764 /var/www/macros/start_vid.sh

## Edit video Boxing behavior
sudo nano /etc/raspimjpeg
# Change MP4Box_cmd command line from:
MP4Box_cmd (set -e;MP4Box -fps %i -add %s %s > /dev/null 2>&1;rm "%s";) &
# to:
MP4Box_cmd (set -e;MP4Box -fps %i -add %s -add /var/www/media/audio.mp3 -delay 2=-100 %s > /dev/null 2>&1;rm "%s";) &
# Be aware that I added -100ms delay to trackID 2 (audio) as my audio is a bit slower than video
# change delay value according your *.mp4 outcome

## Give permission to RPI_Cam_Web_Interface to call follow commands
sudo nano /etc/sudoers.d/RPI_Cam_Web_Interface
# Add follow lines to it
www-data ALL=/usr/bin/arecord
www-data ALL=NOPASSWD: /usr/bin/arecord
www-data ALL=/usr/bin/arecord
www-data ALL=NOPASSWD: /usr/bin/arecord
www-data ALL=/usr/bin/aplay
www-data ALL=NOPASSWD: /usr/bin/aplay
www-data ALL=/usr/bin/pkill
www-data ALL=NOPASSWD: /usr/bin/pkill
www-data ALL=/usr/bin/kill
www-data ALL=NOPASSWD: /usr/bin/kill
www-data ALL=/usr/bin/lame
www-data ALL=NOPASSWD: /usr/bin/lame

## Reboot system
sudo reboot

roberttidey, 2021-09-13:
Thanks for your work on this.

This would be good to add to the wiki for others who would be interested.