silvanmelchior/RPi_Cam_Web_Interface #635: How to change default arrangment in preview page?

markxyx, 2021-09-15:
Now the way it sorts videos ascendly by default in the preview page. We can change it to descending by manual. But every time the system reboot will change it back to ascending. 
How do I make it sorting videos descendly by default?

roberttidey, 2021-09-15:
The filter type and sort arrangements are persisted in cookies so that they are tied to the browser client. 

These should not be affected by reboots etc of the Pi and a quick test on one of my systems showed that the sort order is kept after a reboot.

Check that you are allowing cookies.

markxyx, 2021-09-15:
> The filter type and sort arrangements are persisted in cookies so that they are tied to the browser client.
> 
> These should not be affected by reboots etc of the Pi and a quick test on one of my systems showed that the sort order is kept after a reboot.
> 
> Check that you are allowing cookies.

Thanks a lot. It is my cookies setting mess this thing up.

closed 2021-09-15 by markxyx