silvanmelchior/RPi_Cam_Web_Interface #636: Multiview not working for me

nschertz, 2021-09-26:
Hello -

I have the RasPi setup and working for a single feed, but tried to use multiview.html without success. In /var/www/html/ , I copied the multiview.jsonD to multiview.json. I then changed the IPs in multiview.json for each source to the IP for the working camera.  I attempted to access multiview.html using CameraIP/html/multiview.html substituting the actual camera IP for CameraIP.  I don't get the 4 feeds expected.  In fact, I don't get any feeds. If I revert back to the single feed all is well.  Thoughts?

Thx Norb

nschertz, 2021-09-27:
Update.  I noticed the ownership of the copied multiview file was not set to www-data, so I used chown to change both the owner and group to www-data.  Still no luck.  I even reduced the number of hosts to 1 in the new multiview.json file to see if I could get multiview.html to work with only one camera.  No Luck. when I simply type in CameraIP/html, camera feed worked as expected.

roberttidey, 2021-09-27:
I suspect the issue is with the json config file.

This defines the host url. The example given is for the case where no subfolder is used, e.g. the server files are at /var/www
If a subfolder is used then it should be included in the host definition (see wiki).

As you are using the html subfolder then the hosts in the json file should be http://ip/html/

If that does not fix the problem then please zip your multiview.html and multiview.json file and attach here.

nschertz, 2021-09-27:
Robert -
Thank you so much!!  That fixed the issue. Knew it had to be something simple..  

Now that I have it working, I noticed a specific setting I changed in "system" (Style = night) for the individual camera did not prevail in multiview. Is this also true for camera settings? Understand, from the wiki, that multiview only has the ability to display (i.e. all other functionality and setting change options aren't present).
It would be nice if one could toggle between being able to change settings (advanced mode) and merely display ( normal mode ) for each camera in multiview.  I know.. pretty complex.  At the very least, it would be good if camera settings are preserved for each camera when displaying in multimode.  Are you still actively adding new features in this project?
I have been in a fairly extensive search over the past year for a good Rpi live streamer.  Experimented with motion, motioneyeOS, raspvid... long list.  Your approach is the only one I've found that has low latency on a Pi zero W.  Thanks very much for bringing this about. 

 I'm a model railroader who will be using this software with two tunnel cams that will monitor trains in the two 6 ft tunnels I have on my layout.  I have, more than once crashed into a train derailed in the tunnels.implimenting tunnel cams will fix this.  

Thanks again for the help!!
Norb

roberttidey, 2021-09-28:
One  way to expand the multiview is to add url to each image area that takes it through to the camera settings for that camera when you click on the image area. You can then make any changes you want and return back to multiview using the back button on the browser. The camera settings should be preserved.

The easy manual way to do this is to enclose each <img> tag inside a <a> tag with a href property pointing directly at the url for that camera.

So 
<img id="mjpeg_dest0"  style="width: 49%"/>
becomes
<a href="http://192.168.0.100/html"><img id="mjpeg_dest0"  style="width: 49%"/></a>

The more advanced option to achieve the same thing is to modify multivew.php with an extra parameter so that the href in the <a> tags becomes generic and the host reference is applied at the mutlivew.php end using the data from the json file.

Styling isn't supported in multiview as I thought the screen would be pretty much fully occupied wit the images. You can add styling to it by including a css section in multiview.html to do things like setting the overall background colour to black.

I do occasionally add extra features in when somebody makes a request that seems generally useful and I can implement it in a way that doesn't break backwards compatibility.

nschertz, 2021-09-28:
Thanks for the ideas. I may experiment with these when I get a chance. The multiview now works great for my model train tunnel cams. Thanks again for the assistanceNorbSent from my Galaxy

The easy manual way to do this is to enclose each  tag inside a  tag with a href property pointing directly at the url for that camera.
So

becomes

The more advanced option to achieve the same thing is to modify multivew.php with an extra parameter so that the href in the  tags becomes generic and the host reference is applied at the mutlivew.php end using the data from the json file.

roberttidey, 2021-09-28:
OK.

I have now actually added the click through camera settings functionality to the multiview.html and multiview.php in the latest version.

nschertz, 2021-09-29:
Is this available on the git hub site ?Sent from my Galaxy

roberttidey, 2021-09-29:
Yes.