silvanmelchior/RPi_Cam_Web_Interface #641: Update installer.sh to fully support PHP 7.4?

wn-dev, 2021-11-15:
The `installer.sh` script has a few parts that don't fully support specifying 7.4 as the version of PHP, even though that's the version for Raspberry Pi OS Bullseye.  These are the parts of the script I've noticed this issue:

1. The installer UI doesn't include `bullseye 7.4` in the list `stretch 7.0,buster 7.3` shown for the `php` field.
2. In `fn_nginx()`, the script doesn't replace the `/var/run/php5-fpm.sock` value in the `rpicam` file with the correct value for PHP 7.4 `/run/php/php7.4-fpm.sock`.

roberttidey, 2021-11-15:
1. When run on bullseye the php default is set to 7.4. The comment becomes a reminder of what to use for previous OS.
2. Thanks for pointing that out. Corrected now.

wn-dev, 2021-11-16:
Just tried the updated `install.sh` script on Bullseye, but the installer UI (Configuration Options) still defaults to showing `7.3` instead of `7.4` for the `php` field.  Looks like it's because the script is hardcoded to write `7.3` in the `config.txt` file it generates:
https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/dc0fe1658f1ba0c812a4650ac691b1c7bbbad531/install.sh#L81

roberttidey, 2021-11-16:
Thanks. Corrected now.

wn-dev, 2021-11-17:
Can confirm it works.  Thank you.

closed 2021-11-17 by wn-dev