silvanmelchior/RPi_Cam_Web_Interface #643: mmal_vc_component_create failed to create  'vc.ril.camera'

khaosgrille, 2021-11-29:
Hi, 
here is all data i think might be related to the issue
Device: Raspberry Pi Zero W 2
OS: Raspbian 32 bit, updated
Uname -a
```
Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
```
vcgencmd vcos version
```
Oct 29 2021 10:50:02
Copyright (c) 2011 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean)
host buildbot
```
libcamera-hello --version
```
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:17)
libcamera build: v0.0.0+3156-f4070274
```
Camera Module:
Zero Cam Noir (Not sure if an official device to be honest. it is sold at all the official resellers. for example berrybase.de
it uses the  OV5647-Chip like the Raspberry Camera Version 1

Camera works fine with libcamera-jpeg

Initial Error
```
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
sh: 1: /var/www/html/macros/error_hard.sh: Permission denied
```
Fixed the permission errror for now with chmod 777 -R /var/www/html/macros/ 

Afterwards the error of ./start.sh is

```
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
```
Thank you for your help and the program!

GeekyTim, 2021-11-29:
Currently, the Raspberry Pi Zero 2 cannot use any of the Pi cameras. We are
awaiting driver development.

*Thanks*

*Tim Richardson*
*Twitter: @Geeky_Tim*

khaosgrille, 2021-11-29:
oh ok, 

libcamera works fine though and i reinstalled everything and it worked once before rebooting. afterwards i got the error again.
i tried 
"raspimjpeg start"

now i am in a boot loop
is there an alternative to this project to stream video and audio over wifi?

roberttidey, 2021-11-29:
If this a Bullseye install or has been upgraded to Bullseye the RPi_Cam_Web_Interface will only work when the legacy camera stack is enabled. The install.sh should have done this but today I discovered a bug that meant this wasn't fully done.

The latest version fixes this.

khaosgrille, 2021-11-29:
@roberttidey 
I did try this after commit  35f295a (https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/35f295a40c545bce3af4a5a36bfda7242cb884eb).
Btw i had to remove the gpu_mem=256 line because the raspberry pi zero w 2 seems not to have enough memory for libcamera-hello than
https://github.com/raspberrypi/libcamera-apps/issues/166

I do use the version based on bullseye (https://downloads.raspberrypi.org/raspios_lite_armhf/release_notes.txt)

khaosgrille, 2021-12-12:
Hi,
@roberttidey 
did try again and gpu_mem=128 seems not do to the trick. with  a fresh raspbian full (not lite). i got it to work and it survived a reboot, but only after i removed the gpu_mem line completly. Will keep you updated if the problem or the bootloop comes back. If you think removing the gpu_mem line is fine I would probably close the issue for now

theTJ, 2022-01-05:
> The latest version fixes this.

hmm... i've got bullzeye on a Raspi Zero (1) W and have a similar problem, after I tried with "RaspiOS Legacy" and had a problem with another system (I2C Sensor)
Then I read here (not this post but on the wiki-page) that the install.sh checks for the new system and fixes it when you reboot after install...

as it didn't seem to work i changed the config via raspi_config to use the legacy-cam-stack myself but it didn't help either.

And with the idea to "fix" the requested permissions like @khaosgrille descriped, it kicked me out of my ssh and started a bootloop 😞 

The only thing I see now is starting from scratch but this time with the RaspberryPi OS full vesion?!
switching to legacy-cam and trying to install then?

Any better Ideas?
Maybe I try again with RaspiOS Buster instead...

any ideas?

roberttidey, 2022-01-05:
This was originally tested on the medium Bullseye (Desktop, but not full) and the install.sh worked as intended and enabled legacy camera support and allowed things to run normally.

As per the wiki the raspi-config was used to enable camera support before the install was run.

Your comment on enabling legacy camera support via raspi-config was new to me as the original bullseye install didn't have this (just the regular enable camera). I updated raspi-config and the legacy support option appeared in place of the original enable camera option. They must have changed that recently to better support users who had camera software they wanted to continue to use.

I ran this option on a working Bullseye with the camera working OK. I checked the /boot/config.txt and it had added a further start_X=1 line in and had removed the camera_auto_detect line whereas I had commented that one out. After the reboot the camera was still working OK.

However, I have yet to test what happens on a completely fresh Bullseye with just enabling the legacy camera support prior to running the install. It is possible it makes some of the changes I put into switch on legacy support redundant. I'll report back when I have had a chance to do some further tests.

Using Buster is obviously possible to avoid these complications as that natively supports the raspberry camera without having to fiddle with legacy options.

roberttidey, 2022-01-06:
I started with fresh image of BullsEye (30 Oct 2021 Desktop). Did normal raspi-config stuff (expand filing system ,password etc.).

raspi-config Interface options showed enable camera support. I did not enable.
updated raspi-config and then under Interface I enabled legacy camera support.

/boot/config.txt showed that the modifications done by install to this file were already there.

I did normal apt-get update etc then git cloned the RPi Cam Web software. I edited install.sh to comment out the 3 lines doing the config.txt mods.

I then ran the install.sh and all proceeded normally and I ended up with a working camera system.

I don't think commenting out those lines should have made any significant difference but I haven't tested that. I have amended the github to have those changes, and have changed the wiki a bit to reflect this.

theTJ, 2022-01-07:
OK, I found an error in my installation...
I typed "lightttp" instead of "lighttpd" so it wasn't installed.
at least part of this error is gone now and I get a different one 🤦 

"Could not create camera" - I'll take a closer look in there...

roberttidey, 2022-01-07:
OK.

Note that I use Apache myself and that is the main one I test with. Support for other web servers is dependent on contributions from other users that I integrate in. Lighttpd was working OK on Buster but I don't know if anybody has tried it on Bulllseye.

Having said that. a reported error like "Could not create camera" should not be anything to do with web server selected.

theTJ, 2022-01-07:
OK, it clearly has nothing to do with the choice of the webserver.
And also I'm back at the point where this issue startet:
`mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
sh: 1: /var/www/html/macros/error_hard.sh: Permission denied`

A point that I saw now after staring on the error message for quite some time ist that the initial "chmod 777" couldn't help at all because the error is about the first line of the error_hard.sh-script...

> `#!/bin/bash
sudo shutdown -r now`

looks a bit like the initiator of the boot loop:

- autostart of the cam
- the cam throws an "error_hard" and invokes this script
- that script then reboots the system
- the cam is (auto)started again

🤔

roberttidey, 2022-01-07:
I have done multiple Bullseye installs over the last few weeks so I am pretty confident there is no fundamental issue here but I will try doing another test on a Pi Zero (1)

Of course, one possibility is that the camera itself is not being recognised. That will also trigger the error reported. This can happen if the power supply is marginal or the camera cable is not making a correct connection.  It is worth checking both ends are well inserted and also the short ribbon connector on the camera module can sometimes come loose and needs to be pressed down.

Ideally with this type of issue one should try the camera independently. Previously this would have been done with raspivid but that is not present on bullseye and one would have to remove the legacy stack and use the libcamera equivalent.

Perhaps the easier route is to start with a trial Buster OS, check out raspivid, then do a RPi Web Cam install on that to ensure that all the hardware is OK.i

roberttidey, 2022-01-07:
I just took a SD card with RPi Cam out of a Raspberry 3 and plugged it into a Pi Zero 1 and it all booted up fine with Web Cam software running fine.

roberttidey, 2022-01-08:
Just a further double check. I started with raw Bullseye on the Zero 1 and did the install there on the Zero. All good with working RPi Cam.

theTJ, 2022-01-12:
ok, turns out my Problem was mainly a not fixed cable between Pi and camera.

thanks for your Support, @roberttidey

GenerallyClumsy, 2023-01-10:
I'm also having problems with this. My pi is now stuck in a boot loop after I tried chmod 755 *.* on  /var/www/html/macros/

I'm running the light version of Bullseye (32 bit all updated). I've got a camera that only works with libcam so I've enabled legacy support (Arducam IMX327). Any tips would be greatly appreciated.

roberttidey, 2023-01-10:
The legacy stack only supports a limited range of camera modules and the IMX327 is not one of them.

The only way that is going to work is IF libcam support is added to this software. I know a lot of people want that but it is complex and has some technical barriers as the libcam does not have all of the functionality that the legacy stack has.

GenerallyClumsy, 2023-01-10:
Cheers, I'll watch this space. Any suggestions for similar projects that support libcam? It's a same the IMX327 doesn't support legacy stack.