silvanmelchior/RPi_Cam_Web_Interface #647: Can't save motion settings

taiBsu, 2021-12-28:
Hi, I just cloned the RPi Cam Web Interface repo and tried adjusting some settings for motion detection.
When adjusting any of the motion settings, saving changes takes about 5 seconds and afterwards everything gets back to what it was before. this works for none of the settings, neither framerate nor on_motion_detected.

roberttidey, 2021-12-28:
Whenever you change a setting that makes it different from the default it gets saved in the web server folder e.g. /var/www/html in a file called uconfig. The change is also recorded in the log file which you can see under schedule settings.

This should get updated every time you make a change. It sounds like this is not happening for some reason. Check the folder for the presence of this file. It should be owned by www-data and have read/write permissions. If necessary delete it which will cause all settings to revert to default and allow the software to create it next time a change is made.

Did you make any significant changes during the install process?

taiBsu, 2022-01-07:
@roberttidey thanks a lot for answering. I haven't made any changes during the install process. I'll check the uconfig file and see if it has all permissions needed.