silvanmelchior/RPi_Cam_Web_Interface #648: issue with motion in raspberry pi zero 2 w and Raspbian bullseye

placed87, 2021-12-31:
Hello,
I've cloned git repo into my new raspberry with bullseye and I have the following erros in /var/log/motion/motion.log

[0:motion] [NTC] [ALL] [Dec 31 13:21:31] main: Motion thread 1 restart
[0:motion] [NTC] [ALL] [Dec 31 13:21:31] motion_start_thread: Camera ID: 0 is from /etc/motion/motion.conf
[0:motion] [NTC] [ALL] [Dec 31 13:21:31] motion_start_thread: Camera ID: 0 Camera Name: (null) Service: http:
[1:ml1] [NTC] [ALL] [Dec 31 13:21:31] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] [Dec 31 13:21:31] vid_start: Opening Netcam
[1:ml1] [NTC] [NET] [Dec 31 13:21:31] netcam_start: Network Camera starting for camera ((null))
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_read_first_header: Unrecognized content type
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Error reading first header - re-trying
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_read_first_header: Unrecognized content type
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Error reading first header - re-trying
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_read_first_header: Unrecognized content type
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Error reading first header - re-trying
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_read_first_header: Unrecognized content type
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Error reading first header - re-trying
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_read_first_header: Unrecognized content type
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Error reading first header - re-trying
[1:ml1] [ERR] [NET] [Dec 31 13:21:31] netcam_http_request: Failed to read first camera header - giving up for now
[1:ml1] [ERR] [VID] [Dec 31 13:21:31] vid_start: Netcam failed to open
[1:ml1] [WRN] [ALL] [Dec 31 13:21:31] motion_init: Could not fetch initial image from camera
[1:ml1] [WRN] [ALL] [Dec 31 13:21:31] motion_init: Motion continues using width and height from config file(s)
[1:ml1] [NTC] [ALL] [Dec 31 13:21:31] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [NTC] [ALL] [Dec 31 13:21:31] image_ring_resize: Resizing pre_capture buffer to 4 items

Could someone help me? 

Thank you

roberttidey, 2021-12-31:
The errors you are getting are from the external motion detect system maybe caused by some changes to the motion process.

Use the internal motion detection system (under camera settings). That works much better and is very efficient.

placed87, 2021-12-31:
How should I do it? I didn’t touch motion in other way

roberttidey, 2021-12-31:
Under camera settings from main page find motion detect mode (near bottom). Select Internal. The motion button disappears and instead you get motion settings bar (like camera settings. The defaults are a good starting point.

placed87, 2022-01-01:
solved, thank you very much!!!

closed 2022-01-01 by placed87