silvanmelchior/RPi_Cam_Web_Interface #649: Running RPI Cam Web Interface on 64 Bit OS

ConorHackett98, 2022-01-15:
Hi all,

I'm new to this area of using cameras with a Raspberry Pi so apologies if the answer is obvious.  I've recently been trying to get the RPi Cam Web Interface working with my Raspberry Pi 4 which is running a 64 bit operating system.

I originally tried running it with 64 bit Bullseye (php: 7.4) and I was able to install the software perfectly using `./install.sh` but when I tried to run `./start.sh` I got the error: `bash: /usr/bin/raspimjpeg: No such file or directory`.

Since then I've tried using the 64 bit Buster OS (php: 7.3) but when I try running `./start.sh`  I get the new set of errors: `bash: raspimjpeg: command not found` and `PHP Warning:  filemtime(): stat failed for /var/www/html/status_mjpeg.txt in /var/www/html/schedule.php on line 717
`

I was wondering if anyone knows how it might be possible to fix these errors or how to run the RPi Cam Web Interface on a 64 bit OS?

Thanks for taking your time to read this.

roberttidey, 2022-01-15:
I have not yet tried running this on 64 bit OS. I always test against the official release currently 30 Oct 2021 Raspberry Pi OS

On the official install a link is put in /usr/bin 
lrwxrwxrwx 1 root root    22 Jan  6 14:20 /usr/bin/raspimjpeg -> /opt/vc/bin/raspimjpeg

So I would check first on your bullseye 64 whether this link exists. If not then first check that /opt/vc/bin exists and contains the raspimjpeg binary and then try adding the link manually to see if that helps.

The other thing to check is whether a browser can access the web page OK. If raspimjpeg is not running then you won't get camera functionality but you should still see the basic pages OK. This would indicate that this part of the software is functional.

roberttidey, 2022-01-15:
I have tried this on 64 bit OS and I get same result.

The raspimjpeg binary is there in /opt/vc/bin as is the link to it in /usr/bin.

So the error message is misleading.

It may be because the binary is 32 bit and that error message is being thrown anyway.

I did try compiling raspimjpeg under this OS but get errors in the heart of the userland code not in my code so I can't progress this further at this stage.

ConorHackett98, 2022-01-16:
I think it might be impossible to get the RPi Cam Web Interface running on a 64bit OS too.  Thank you very much for your help though!

closed 2022-01-16 by ConorHackett98