silvanmelchior/RPi_Cam_Web_Interface #652: Motion event.sh

Nic97, 2022-03-03:
Hi everyone,
i tried to lauch the motion_event.sh to detect motion and notify via telegram bot.
But where i execute the script i have this error:
/var/www/html/macros/motion_event.sh: 21: [: -eq: unexpected operator

Anyone had the same problem?

Thanks

mlambi, 2022-03-03:
No.
I have a bash script that send a Matt meager via mosquito. Can you run the
script from the CLI?

Nic97, 2022-03-04:
This is the script that cause problem:
------------------------------------------------------------
while :
do
 
  #!/bin/bash
  . /var/www/html/macros/configfile.cfg
  
 
  #If you wish to disable notifications, create a file using "touch /home/pi/nomsgfile.lck"
  if [ -f $nomsg ]; then
    logging no true "Message alerting has been disabled, exiting..."
  exit
  fi
 
  
    #Copy the low res snapshot image from memory to a file
    cp /dev/shm/mjpeg/cam.jpg ${webfolder}/cam_new.jpg
  
    #On motion detection start (parameter is 1 for on, 0 for off)
    fileparams=$1
    if  [ $fileparams -eq 1 ]
    then 
  
    #Send both a single text message and a photo because sometimes the photo message times out if the Telegram servers are busy
    logging yes false "Motion detected from ${cameraName}"
 
    curl -s -X POST \
      ${tokenurl}/sendPhoto \
      -F chat_id=${chatid} \
      -F photo="@${webfolder}/cam_new.jpg" \
      -F caption="Motion detected on ${cameraName}" >> ${curlout} &
  fi
sleep 1m
done
------------------------------------------------------

The link of project is: https://www.moreware.org/wp/blog/2022/03/03/raspberry-pi-cam-bot-telegram-che-rileva-movimenti/

Thanks in advance