silvanmelchior/RPi_Cam_Web_Interface #654: Make annotation size relative to image resolution

s12wu, 2022-04-02:
If you set the resolution of images and videos to different values (e.g. 1440x1080 vs 2592x1944), the annotation will appear larger on the videos than on the images, because the videos have a smaller width, but the annotation itself keeps the same size.
So I would like to have the annotation size relative to the resolution of the capture so that it always appears the same size at different resolutions.
Is that possible?

roberttidey, 2022-04-02:
You can control the text size via the text size control under camera settings. That hasn't been made automatic as different users will want different sizes even for the same resolution.

s12wu, 2022-04-03:
Well, I thought it could be added as a separate option, not necessarily as a replacement for the current annotation size input field