silvanmelchior/RPi_Cam_Web_Interface #656: Help: Modify text of the broadcast

estebansanchez75, 2022-04-06:
Good day, I am doing a project with this live streaming program, I am not very expert in php programming and I have a problem, I want to be able to modify the text of the live broadcast, which says: "RPi Cam 2022 04 05" to be able to run this text on the left and to be able to add a new text on the right, a text that is written by a user, but I don’t know how to do it, if they could guide me in this step it would be very helpful, thank you very much

roberttidey, 2022-04-06:
I assume you mean the annotation text which is superimposed on the video.

The basic information is in the wiki under the naming scheme section https://elinux.org/RPi-Cam-Web-Interface#Naming_Scheme

You configure what text you want to appear by editing the annotation text string under camera settings. The text is made up of fixed characters and variables which are interpreted for every frame of video. So, for example the variable %Y is replaced by the 4 digit year.

Most of the variables are concerned with the parameters of the recording but there is one special one %a which causes the content of a file to be inserted (/dev/shm/mjpeg/user_annotate.txt). The contents of this file may be changed by any other process (like a python program) and will be reflected immediately in the superimposed annotation. A simple of example of this would be to read temperature from a sensor and put the temperature value in the file.

There are two other points. First there is no direct control over positioning; it is top centred. If you want to make it appear to start from the left then you should add spaces at the end to shift it left. Similarly if you add spaces at the start you will shift it right. Second the character set is basic ASCII, Avoid using control characters like CR and LF

estebansanchez75, 2022-04-06:
That is to say that I cannot have two different text files, if I want to print in the video the temperature sensor and in turn another type of sensor as would be the procedure? since only what is inside the /dev/shm/mjpeg/user_annotate.txt file can be placed in the video, how could you place another text file?

fcruz-ehr, 2022-04-07:
I don't see why you couldn't have two programs write their output to user_annotate.txt... Hard to say specifically without knowing what these external things are that you are talking about, but having some outside script take the content of A and the content of B and then write A + B in user_annotate.txt doesn't sound like a big problem, if you already have something working with these sensors or whatever they may be if I understand you correctly

roberttidey, 2022-04-07:
Exactly. Whatever string is in that file will be used so you can compose multiple items in if you want to. But don't make it a mult-line file just concatenate the items in one line.