silvanmelchior/RPi_Cam_Web_Interface #658: Libcamera not supported

azsde, 2022-04-21:
Hello,

I have been using this software for a while now and I got a new camera which has better image quality and has autofocus as well.

However, it seems that the camera only is usable using libcamera (ArduCam IMX519 16Mp Camera)

Do you plan on porting RPi_Cam_Web_Interface to use libcamera someday ?

roberttidey, 2022-04-21:
There are several items making a switch to libcamera difficult as well as the significant effort involved. For example, libcamera does not provide all the facilities that the MMAL interface does including the access to motion vectors that are used to provide efficient motion detection. I also have concerns about the cpu usage in making the switch. The MMAL interface runs very well even on Pi Zero v1 making low cost camera interfaces easy to achieve.

I do continue to review this to see when it may become more feasible but I can't give any time scales. One possible option would be to produce a separate version that was only supported on higher end Pis but that incurs significant maintenance effort.

azsde, 2022-04-21:
How hard would making a version without motion detection be ?

So far my usecase was to be able to start / stop recordings manually while being able to see what the camera is actually filming.

I could obviously simply use the terminal and keep on recording manually, but I didn't find a way to have a live feedback while recording.

roberttidey, 2022-04-21:
It would be a substantial amount of effort to switch from MMAL to libCamera even without motion detection. The internal structure of how all the internal streaming works currently relies on the MMAL plumbing and call back mechanisms. In addition all the camera settings are built around the options that MMAL supports so the web camera settings would need a lot of rework.

Everybody's use case is different but I believe that motion triggering is a common requirement. I have tried successfully so far to keep to one version to satisfy a superset of needs.

Motion triggering can be done by analysing pixel changes between frames instead of using motion vectors and indeed the original software did just that using the external motion detection support (which still exists). But I found this was much more cpu intensive and also did not give such good results. The cpu part would not be a problem on higher end Pis which is why I said that one option would be to produce a libCamera version that was only supported on higher performance Pis. This would probably be Ok on Pi0v2

I do realise, of course, that the MMAL method is now a dead end as the Raspberry camera support has moved to libCamera, so if this software is to continue then something will be needed, but there is only one developer (me) and I have lots of other projects as well. Source code is available for all parts and I welcome any effort to make changes or improvements.

azsde, 2023-01-09:
Hello,

As of today, the new official raspberry camera V3 has been released, they are not compatible with the old camera stack and only are compatible with libCamera.

I assume they will stop the production of the old V2 camera so it might be mandatory to update this software in the future to add compatibility with libcamera.

kaihokori, 2024-10-26:
Hi @azsde, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). Be aware that not all functionality has been implemented but it should be a good starting point ^_^

azsde, 2024-10-26:
> Hi @azsde, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). Be aware that not all functionality has been implemented but it should be a good starting point ^_^

That's awesome !

I also started my own version that also uses Libcamera2 and I'll probably release it when it will be more stable !