silvanmelchior/RPi_Cam_Web_Interface #660: Feature request to have a static link to the latest timelapse photo

La-te, 2022-04-27:
I would like to easily access the latest photo captured by timelapse operation from a static URL like /media/latest.jpg. Would that feature be possible to have built-in? I could of course solve this in many ways just with shell scripting, web server etc. However, I believe some others might benefit from this and I expect it to be rather simple implementation.

I know there is cam_pic.php but it is having the dimensions of the video stream which I'm setting lower than for timelapse photos. I would like to have an easy way to see and automatically fetch the latest timelapse shot as I'm running the timelapse constantly with certain interval.

roberttidey, 2022-04-30:
I am looking into providing a php url that will return the latest time lapse.

roberttidey, 2022-05-01:
Using the url /cam_picLatestTL.php will return the jpeg of the latest time lapse images