silvanmelchior/RPi_Cam_Web_Interface #662: Settings/Parameters for asking storage of images and viseos on an USB memory key connected on the Raspberry

pdemeure, 2022-05-03:
Insead of saving images and vidos on the micro SD, I'd like to save them on a USB emory key. I'd like to do that because my raspberry is included and attached in an outdoor case, and it is much more easy to remve a USB key than the micro SD when I want do transfer images to my PC.
Is there an easy way to set the storage on a USB memory key instead of the standard place ?
I'd like to be able to set that easyly without changing folders, files, configs and so on.  It would be great to be able to change storage forlder path i camera or system settings, in order to be able to reverse on standard storage in case of need.
Thanks in advance if you can help !
Kind regards,
Pierre

roberttidey, 2022-05-03:
Many people do store their media elsewhere, either on usb memory drives, real USB disk drives or on mounted network drives.

The settings file in /etc/raspimjpeg does allow the locations of media files to be moved elsewhere but you do have to be a bit careful to avoid disturbing the normal web operation.

Instead the better way to do this is to leave the settings as they are and instead use Unix link facility to relocate the media folder to whichever storage location you want to use as mentioned in the FAQ section of the wiki. This is not only easier but also avoids complications when upgrading the software. You can either use the ln command to create a symlink from your external folder back to media or use fstab to mount the external folder on top of media.

One extra point is that if you choose to use a network drive then there is a facility called a boxing folder (described in Wiki under CIFS / Samba / Windows Share) which avoids problems caused by extra delays that can occur when storing files over a network.

pdemeure, 2022-05-04:
Hi,
My USK key is automatically detected. With fsik command I get the hereunder infos 👍 Disk /dev/mmcblk0: 59.2 GiB, 63596134400 bytes, 124211200 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x332df385

Device         Boot  Start       End   Sectors  Size Id Type
/dev/mmcblk0p1        8192    532479    524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 124211199 123678720   59G 83 Linux

Disk /dev/sda: 57.9 GiB, 62109253632 bytes, 121307136 sectors
Disk model: Ultra USB 3.0
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start       End   Sectors  Size Id Type
/dev/sda1          32 121307135 121307104 57.9G  c W95 FAT32 (LBA)

Could you ne kind and give me infos on the command or file creation I have to perform in order to make the best way like you explain in your above comment ?
Thanks in advance
Pierre

roberttidey, 2022-05-04:
This is a good guide to mount the usb drive.

https://raspberrytips.com/mount-usb-drive-raspberry-pi/

Note in particular.

1. You want to use the automatic method (e.g fstab) and using the UUID as described.
2. You want the mount point to be the media folder. e.g. /var/www/html/media not the /mnt/usb used in the guide.
3. You don't need to create the media folder as it already exists but make sure it is empty first.
4. You can skip the update stuff as your system is new anyway.

After reboot you can test to make sure files get stored there OK.

pdemeure, 2022-05-04:
Hi,
I followed your instructions. However, after creation of the fstab and addition to the end of the line UUID=xxxxx  after reboot, the Raspberry doesn't connect at all. My  Putty connection stays on black screen and after a while I get an error message about network connection not responding.
What about that ?
I now have to re-create a complete system ?
(I connect to the raspberry via WiFi)

roberttidey, 2022-05-04:
An incorrect fstab can cause boot problems. Adding nofail in the fstab options can help avoid getting stuck.

When faced with a problem like this there are 2 basic methods of recovery to avoid starting again.

First you can remove the SD card, put it in a USB card reader stick and insert it into another linux system (another raspberry or a PC with linux boot disk. This will give you access to the /etc/fstab file to edit.

Second you can remove the SD card put it in USB reader stick and insert it into a Windows PC. You can only see the boot partition but you can Append init=/bin/sh at the end of cmdline.txt  When you reboot you will be in a root shell. You need to mount the filing system as read/write by running mount -n -o remount,rw /     You can then edit the fstab file.

pdemeure, 2022-05-04:
I've followed your instruction by adding init=/bin/sh at the end of cmdline.txt, but after reboot I still not get network connection. It seems that fstab file breaks the network wifi connection...
If I manually mount the USB drive, it works correctly. Then it is the fstab which makes the issue.
My fstab addition line is this one 
UUID=1208-3D5F /var/www/html/media vfat uid=pi,gid=pi 0 0   (I also tried wit ext4 instead of vfat but nothing better)
I exactly followed instruction from the url you send me.
My only solution is to re-make a clear install of RCWI...

pdemeure, 2022-05-04:
I'll try again with PARTUUID. The USB drive was formatted with option format of Pi manager.

Also something strange in the behaviour : if I mount manually the usb key, I can write on it without issue. For instance, I can copy a file from a RPI folder to /var.www.html/media. After umount of the key, I can see the copied file on my PC.

BUT !   The application can't record images ! The button "record image" does nothing and no image are saved on the USB key correctly mounted...

roberttidey, 2022-05-04:
That will be a permissions issue. You need to use chown to change the ownership of the mounted drive to www-data and also chmod to set read write permissions of the volume to say 666

roberttidey, 2022-05-04:
Have you still got that extra item in cmdLine.txt? The commands seem to indicate you are maybe logged in as root rather than pi.

I would remove that if so.

I would use sudo chown www-data:www-data  /dev/sda1

This ensures both group and user are set correctly.

Use ls -l on both /var/www/html and /dev/ to check that ownership and permissions are set OK on /var/www/html/media and /dev/sda1

pdemeure, 2022-05-04:
I've checked all all along today...
I never succeed giving permissions on /var/www/html/media
've never been able to give ownership on /var/www/html/media to www-data user. I've never been able to give permissions on this directory.
I've nothing as extra item in cmdline.txt because I re-created a clear new os.
media always belongs to root... and impossible to modify permissions or ownership.

-rw-r--r-- 1 www-data www-data 25066 May  4 14:44 loading.jpg
drwxr-xr-x 2 www-data www-data  4096 May  4 14:44 macros
**drwxr-xr-x 3 root     root     32768 Jan  1  1970 media**
-rw-r--r-- 1 www-data www-data   346 May  4 14:44 min.php
-rw-r--r-- 1 www-data www-data  5966 May  4 14:44 motion.php

roberttidey, 2022-05-05:
OK. Got  a chance to do this myself. This is what worked for me with a 32GB FAT32 stick

1. Let raspberry start up and automount drive. This will typically be at /media/pi/[diskLabel]
2. Get UUID of drive
3. edit /etc/fstab and add
4. UUID=XXXX-XXXX /var/www/html/media vfat defaults,auto,uid=www-data,gid=www-data,rw,nofail,noatime 0 0
5. unmount existing automounted drive sudo umount /dev/sda1
6. Refresh mounts from fstab   sudo mount -a

The camera now recorded to the stick and the preview page showed free memory as 32GB

Reboots should work OK.

pdemeure, 2022-05-05:
Hi,
1- When Raspberry starts, I have no automount drive. And no /media/pi/[diskLabel]
2- Then I've not go further.
3 - I succeed mounting the USK key drive with the following command :
     sudo mount /dev/sdb1 /var/www/html/media -o umask=000
4 - Root is still owner, but permissions are all open ,  and RPI_Cam_Web_Interface can write images and videos without any issue !
5- Next step : I'll try to make automount with fstab....
     UUID=XXXX-XXXX /var/www/html/media auto user,umask=000,utf8, 0 0

pdemeure, 2022-05-05:
Unfortunately, with automount method using UUID, yhe same issue is still there. Raspberry pi doesn't start and is not seen in network. Is there a conflict with fstab and wifi or network ?
I give up....
I think that I'll make manual mount that works...

roberttidey, 2022-05-05:
The tests I did were on a wifi connected raspberry and I just used putty to access it.

If there is no entry in fstab then the standard raspbian OS will automount the drive at the default mount point. A df command should show that. That is where I started from.

pdemeure, 2022-05-06:
Here is a copu of result from df -h  

pi@raspberrypi0:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  1.7G   55G   3% /
devtmpfs        404M     0  404M   0% /dev
tmpfs           436M   24K  436M   1% /dev/shm
tmpfs           436M  5.9M  430M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           436M     0  436M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   49M  204M  20% /boot
tmpfs            88M     0   88M   0% /run/user/33
tmpfs            88M     0   88M   0% /run/user/1000

That don't show defaut mount point for the USB Key  (Sandisk)

pdemeure, 2022-05-06:
Another question 
 I'd like to use a RTC module in order to keep date/time. How to use it in System settings ? Do I have to install something or is RPI_Cam_Web_Interface able to use the infos from the RTC ?

roberttidey, 2022-05-06:
I don't know why you can't get the USB to automount. The only thing I can suggest is that starting from a clean Raspian OS then insert the USB drive before installing any software and check that automounts.

The main purpose of a RTC module is to maintain the date/time on raspberries which do not have access to the internet time servers where they would normally sync their clock. 

You do need to do stuff to both make an RTC work and also allow the Raspberry to use it for its main time source. Check out the various tutorials on how to do this e.g.

https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi

Nothing special needs to be done on the RPi_Cam side. It will use the date/time from the raspberry irrespective of what has sync'd it.

roberttidey, 2022-05-06:
OK. The automount is associated with the desktop environment. If there is no local monitor (e.g. headless) then it doesn't kick in.

Although most of my cameras are headless the one I tested with had a local lcd screen so that is why mine automounted.

So, if you just do the steps but ignore the instruction to unmount the automounted drive then it should work.

pdemeure, 2022-05-10:
Hi Robert,
I now have a correct and good working configuation.
I made a clean SD using Buster with Desktop. 
When I want to use a USB Key, I unmount it from initial automatic mount, and I mount it manually with this instruction 
**sudo mount /dev/sda1 /var/www/html/media -o umask=000**
That mounts the key in the correct directory with permissions to write on it.
Thank you again for your precious help.
Pierre