silvanmelchior/RPi_Cam_Web_Interface #664: Use of a better camera than raspberry V2

pdemeure, 2022-06-02:
Hi,
I'm disappointed by the poor quality of the v2 raspberry camera. Not enough sharp even wit focus adjusted, not very good treatment of colours and so on.
Then I d like to find another camera with better features. Have experimented good cameras that could replace the V2 , connect directly on the raspberry cam port/socket and work immediately without need of drivers or config ?
Thanks in advance
Pierre

roberttidey, 2022-06-03:
Currently, this software can only be used by native Raspberry cameras supported directly by the GPU.

Bullseye OS does support different cameras but using a different camera interface (libCamera). At the moment there are no plans to switch to using this method as there are some features lacking which makes the port difficult.

pdemeure, 2022-06-03:
Hi,
I just saw in the wiki that there are some camera usable, other than V2 or V1. Is there one which coulb be much better than V1 or V2 ? 
However, V2 works but is difficult to set up at correct focus. It is enogh for observation, but for improved use, I have to optimize the videos with a video editor for dehaze, sharpening, and so on.

roberttidey, 2022-06-03:
There are versions of the camera which are based on the standard V2 but have different lenses and manually adjustable focus adjustable focus.

mlambi, 2022-06-03:
Most of the aftermarket cameras are the 5MP version, and the official v2 with 8MP does not have any/many aftermarket versions with IR cut-filters, c-mount lenses, etc.
However, there is the HQ cam, at 12MP and a variety of lenses available. Much more expensive, but an interesting option.

pdemeure, 2022-06-03:
Hello,
I've seen this 12mp camera and sets of lenses. It is much more expensive but if the quality is really much better that can be an option.
BUT are you sure that it can be used directly by RCWI without any driver, just by plug and play on the camera socket of a raspberry / Buster ?

pdemeure, 2022-06-05:
> 

Could you give me the references of these advanced cameras but based on V2 and which could works with RCWI on raspbessry running Buster ?
Thanks

roberttidey, 2022-06-05:
Note that I don't have any direct experience of cameras beyond v2 so this is just based on what I believe to be true.

You can use camera using the imx477 sony camera chip as used in the HQ camera. You can either use the official raspberry one or several alternatives based on this chip. Some have replaceable lenses and focussing options.

Just search for raspberry cameras using the imx477camera chip.

m-sja, 2022-06-07:
I do own the raspberry HQ cam.
I did not test the HQ Camera with RPi_Cam_Web_Interface yet, but once legacy-camera is activated (i'm using bullseye) i'm expecting it to work fine (except RPi_Cam_Web_Interface does not support all of the new fancy features of this camera yet).

With good lenses the quality of the HQ Camera is very nice! There are (cheap) lenses available for different usecases.
I like to use (cheap 5MP) s-mount (M12) Lenses with a cs to M12 adapter. But correct lens selection is not trivial (imho).

Please see this [Thread](https://forums.raspberrypi.com/viewtopic.php?t=272658l) with a lot of information and sample shots regarding the HQ Camera.

One note:
There are other camera modules using the IMX219 and IMX477 offered to work with "jetson" boards. These camera modules do not work on raspberry pi (except maybe some compute modules) due to driver licensing using a crypto chip on the official camera modules.
The 3rd party Models with OV5647 chip are not affected by this limitation.
There are models (some with automatic ir-cut filter) where you can use s-mount (M12) or cs/c-mount lenses.
Maybe this is a option for you.

m-sja, 2022-06-10:
Hi, i just checked the RPi HQ-Cam on my CM4 running bullseye (installed via rpi-imager).
(The only problems i ran into were related to my new CM4 (compute module) + 3rd party carrier board i was not familiar before)

RPi_Cam_Web_Interface runs fine with HQ Cam and bullseye!

1. Enable the legacy camera stack using "sudo raspi config" (menu Interface Options -> Legacy Camera)
2. reboot
3. install RCWI according to wiki
4. use it
I did no in-depth testing of any features of RCWI, just installed it, opened the Website in Browser, clicked Motion-detection-start, waved my hand over the camera an it recorded a short clip.

Just as simple as that.
Well done!!! Thats how software should be :-)

pdemeure, 2022-06-10:
Hi,
Please could you provide a link to this HQ camara (to amazon or othe seller). That will be useful in order to be sure of the good product ?
Thank you for your test.
Pierre

m-sja, 2022-06-10:
Here is the Link to the [official product page](https://www.raspberrypi.com/products/raspberry-pi-high-quality-camera/) .
If you scroll down, there are distributors listed for different regions in the world.

The camera modules does not come with a lens!
RPf offers two "official" lenses, but there are a lot more compatible (and incompatible) to chose from.
Before ordering i would recommend to read some information about
- focal length
- sensor size
- fish eye vs pinhole
- s-mount, cs-mount, c-mount
I can recommend this [site](https://www.arducam.com/docs/lens/introduction/) they have a lot of good and information

cs and c is the same thread-size, the only difference ist, that the distance between lens and sensor ist shorter on cs.
with the HQ cam module you get an adapter which is used to screw c-mount lenses to the cs-mount on the module.

Btw in RCWI you have to set still image res. to 4056x3040px to get the full 12MP quality.
Also it is important to focus the lens well. I do this with libcamera before switching to legacy camera:
`libcamera-hello -t 0 --info-text "%focus" -roi 0.45,0.45,0.1,0.1`
This displays a live video on desktop.
The title bar contains live focus metric (higher number is better).
The roi (region of interest) is like digital zoom which makes it more easy to focus well.

I wish you good success with lens selection and your project!

pdemeure, 2022-06-11:
Hi,
A great and warm Thank you for these infos !
Pierre