silvanmelchior/RPi_Cam_Web_Interface #667: Permission issue creating /var/log/motion/motion.log

m-sja, 2022-06-07:
First of all a big Thank You for this nice project!
I am not sure if this is an issue of RPi_Cam_Web_Interface, but as there is an installer as well as good documentation, i think this issue can be addressed here:
I started from a clean install of "Raspberry Pi OS Lite (32-bit)" (=Bullseye) using Raspberry Pi Imager.
I followed the instruction on the wiki site which worked flawlessly until i got stuck when trying to enable motion detection in the web interface.
Motion service exited with error message of denied permission to create the log file.

- As suggested i checked the file: /etc/motion/motion.conf
   (file was motion:www-data 664) and changed it to www-data:www-data 664, but that did not solve the issue.
- I was not able to check the process-owner of motion as the process exited immediately
   (i'm sure there is a way, but not within my current linux skills)
- The log file did not exist at this point, nor did the folder within the log directory
   (mentioned [here](https://raspberrypi.stackexchange.com/questions/78715/motion-daemon-var-log-motion-motion-log-permission-denied) )

Here is the solution which worked for me:
I created an empty folder: sudo mkdir /var/log/motion
Then changed the owner: sudo chown www-data:www-data /var/log/motion

The motion process (once started via web-interface) is now able the create the logfile due to the correct permissions.
But on every reboot i get the error message again when executing: systemctl status motion.service
It seems as the default user for motion.service is "motion", but when started via the web interface the user is "www-data"

I do not have the answer how to fix this issue, but i can think of two slight improvements:
1. Add this folder and set the correct permissions within the installer when detecting bullseye
    (maybe this is more a motion-problem and not a RPi_Cam_Web_Interface-problem and should be solved elsewhere)
2. At least add this workaround to the wiki Page for bullseye os to prevent new users to run into this issue

Best regards
Marian

roberttidey, 2022-06-08:
The problem is with the external motion program which changes around from time to time.

Use the internal motion detection (under camera settings). This works much better anyway, and uses much less resources as it is based on the motion vectors available from the camera system.

I have changed the default to be internal detection to avoid this.

m-sja, 2022-06-08:
Hi @roberttidey ,
thanks, it is a good solution do set the recommended setting as default!
If someone is running into the same issues as i did, one should be able to find this issue and my solution using the github-search.

Thanks for your quick response!

closed 2022-06-08 by m-sja