silvanmelchior/RPi_Cam_Web_Interface #669: BULLSEYE, Error in Android when connecting web

abejarutas, 2022-06-24:
I upgraded my Bullseye lite OS and then installed RCWI 6626 following all instructions from https://elinux.org/RPi-Cam-Web-Interface.
With the desktop PC, and with the Buster system cameras it works perfectly, being able to connect from any Andriod or laptop.

But with the camera upgraded to Bullseye, the old 6624 appears on top, the buttons don't activate and it doesn't work at all.
Somebody could help me?
I'm new and I don't know how to explain myself better, my apologies.
J.Serrano

roberttidey, 2022-06-24:
What do you mean by upgraded Bullseye? 

A clean install of Bullseye followed by install of RCWI does give a working installation using the legacy camera support option. I can't see how that process could show a version other than 6.6.26

abejarutas, 2022-06-24:
Thanks Robert, you're right.
I use the google translator and that has confused me. OK I have two scenarios:
BUSTER:
A clean install of Buster Lite 32b followed by install of RCWI 6626. Everything always works perfectly: PC, laptop, mobile and Android tablet.
  Pi3A+ and PiZero2W boards.
BULLSEYE:
A clean install of Bullseye Lite 32b followed by install of RCWI 6626 works using the legacy camera support option.
It only works perfectly on PC and laptop.
It does NOT work when connecting from mobile and Android tablet.
But the most curious thing is that version 6624 appears at the top of the website. The buttons do not respond at all, being marked by a black border.
  Pi3A+ and PiZero2W boards.
I'm sorry I can't explain myself better after 20 or more tests on both cameras, that's why I've come to ask for help.
Thanks for your time.
  Note: (I have been using the different versions of RCWI for years, no problem)

roberttidey, 2022-06-24:
Thanks for extra explanation.

What version number appears when accessing the bullseye from PC and laptop?

If the version number is OK on those then I suspect that something might be being cached in the android / tablet browser. If you can clean the cache that might help.

If the version is wrong on the PC then double check the install by looking at the config.php in the install folder (e.g. /var/www/html). Near the top there should be a line 

define('APP_VERSION', 'v6.6.26');

abejarutas, 2022-06-24:
Thank you very much, robert, I answer in order:
1.- The version that appears on PC or laptop is correct 6.6.26
2.- I will try to find another device that does not have the old version 6.6.24 in cache. This was one of my suspicions.
3.- The version is correct on the PC and it works perfectly and quite well, but even so I have looked at the file in question config.php:

// version string
define( 'APP_VERSION' , 'v6.6.26' );

As you can see, everything is correct, except point 2 about the cache of the devices that I am going to investigate thoroughly. When I have done it, I will give you an answer.
Thank you very much

abejarutas, 2022-06-25:
Solved!!
After nearly three hours!! of trying to clear the hateful Chrome cache on both devices, I ditched Chrome and installed Edge for Android...and when I synced my Microsoft account...magic happened!!
Anyway, in the end what I really didn't understand is why this has only happened to me with bullseye and Chrome .
Now my four cameras are already working perfectly, thanks for your work and advice.
Jesus Serrano

closed 2022-06-25 by abejarutas