silvanmelchior/RPi_Cam_Web_Interface #671: Capture an image every 5 minutes

mfraser, 2022-07-03:
I'm currently using my own Python program to capture an image every 5 minutes constantly at 00,05,10 etc to create a time lapse video of the day.
I would like to use RPi-Cam-Web-Interface as it will give me the option of changing things like white balance depending upon the time of day.
Is it possible to make RPi-Cam-Web-Interface capture an image like this and also have the current image as a separate image?

roberttidey, 2022-07-03:
Capturing a sequence of images at regular intervals is the built in time lapse function of the RPi interface. You just set the time lapse interval to 600 (for 5 minutes and start the time lapse operation.

I'm not sure I understand exactly what you mean by "current image as a separate image". The preview will show the current camera image live. If you wanted a separate copy of the latest time lapse image copied to a separate fixed file then you could do that using a macro which would be a simple bash script.

mfraser, 2022-07-03:
> Capturing a sequence of images at regular intervals is the built in time lapse function of the RPi interface. You just set the time lapse interval to 600 (for 5 minutes and start the time lapse operation.

Trouble is, if I start the time lapse running at say 10.03, then it will capture 5 minutes from then at 10.08, 10.13. I want it so that it will always capture at 00,05,10,15 etc whenever I start the time lapse running.

> I'm not sure I understand exactly what you mean by "current image as a separate image". The preview will show the current camera image live. If you wanted a separate copy of the latest time lapse [...]

If I want to use the current capture elsewhere, I need to know what the current image is.

roberttidey, 2022-07-03:
Use the scheduler to start (and stop) the time lapse at a specific time. Set the scheduler to Fixed Times mode, set time for the start and end as times,  and then enter the start command tl 1 and stop tl 0 in the period start fields.

Two example methods of accessing the current capture. 

1) Use the macro method to copy the current image to a fixed file name like current.jpg

2) Include a command to find the name of the most recent file in the media folder which will be the latest time lapse capture. (e.g. ls -lrt | tail -n 1)