silvanmelchior/RPi_Cam_Web_Interface #673: Vector Preview broken

emjas92, 2022-08-01:
Turning on `vector_preview = 1` breaks rpicam (for me at least).

After enabling Vector Preview in the web interface, it shows the error "Error in RaspiMJPEG: Restart RaspiMJPEG or the whole RPi"
Restarting does not fix it.
Turning off vector preview in the web interface does not work (remains turned on) - I have to edit uconfig and do start.sh

Here's the log.

> {2022/08/02 09:56:20} Change: vector_preview = 1
{2022/08/02 09:56:20} recommended video buffer size 65536
{2022/08/02 09:56:20} h264 size set to 131072
{2022/08/02 09:56:20} recommended video buffers 1
{2022/08/02 09:56:20} h264 buffers set to recommended 1
{2022/08/02 09:56:20} Error: Could not create connection camera -> video converter
{2022/08/02 09:56:20} Set up internal detect width=82 height=61
{2022/08/02 09:56:20} Motion mask /var/www/rpicam/outdoor.pgm loaded
[2022/08/02 09:56:21] Logged in user::
[2022/08/02 09:56:21] UserLevel 6
{2022/08/02 09:56:32} Watchdog detected problem. Stopping{2022/08/02 09:56:32} SIGINT/SIGTERM received, stopping
{2022/08/02 09:56:32} Error: Could not disable video port
{2022/08/02 09:56:37} RaspiMJPEG Version 5.10.2
{2022/08/02 09:56:38} recommended video buffer size 65536
{2022/08/02 09:56:38} h264 size set to 131072
{2022/08/02 09:56:38} recommended video buffers 1
{2022/08/02 09:56:38} h264 buffers set to recommended 1
{2022/08/02 09:56:38} Error: Could not create connection camera -> video converter
{2022/08/02 09:56:38} Set up internal detect width=82 height=61
{2022/08/02 09:56:38} Motion mask /var/www/rpicam/outdoor.pgm loaded
{2022/08/02 09:56:38} Opening FIFO 0 /var/www/rpicam/FIFO 7
{2022/08/02 09:56:38} Opening FIFO 1 /var/www/rpicam/FIFO11 8
{2022/08/02 09:56:38} MJPEG streaming, ready to receive commands
{2022/08/02 09:56:38} send smd 9
{2022/08/02 09:56:38} Starting command loop
[2022/08/02 09:56:38] Reload parameters command requested
[2022/08/02 09:56:38] Scheduler loop is started
[2022/08/02 09:56:39] New period detected 3
[2022/08/02 09:56:39] Send em auto
{2022/08/02 09:56:39} Change: exposure_mode = auto

This has been happening for a while now, though I don't remember when it started - perhaps it was when I switched from Rasbian to DietPi.
I have tried a fresh install of DietPi + RCWI.
I will try out older verions of RCWI to see if it works in a previous release.

roberttidey, 2022-08-02:
I have just checked latest version of RCWI with Raspbian Bullseye and vector preview works for me.

I have not tried with DietPi. Note that with Bullseye one has to use the leggacy camera support. I don't know what happens with DietPi.

emjas92, 2022-08-02:
Thanks for checking.

I have a RPi3 and RPi1 both with DietPi Bullseye, and both have the vector preview issue.
But it's strange that all other RCWI functionality is working, including saving the vectors to the .dat file.

I'm currently troubleshooting my motion detection settings and would be using the vector preview, but am working with the .dat file for now.

BETLOG, 2023-11-30:
Broken here too:

- Normal functioning camera stream.
- I set Motion Vector Preview to on, and the stream becomes a tiny mask-sized image that displays white motion-vector specks. The image is so small its not practical for determining where the mask needs adjustment.
- The on/off selector now shows off, despite vectors being on. Selecting off does nothing. Selecting on re-initializes the motion vectors and contracts the previously expanded Motion Settings panel, the selector remains 'Off'.
- Aside from a reboot; I know of no way to restore the normal camera stream.
- Video recorded by motion detection during this time are normal, except their preview image seems to display what looks like the motion vector preview, or maybe just a very low exposure gain moment as the camera switches.

`Linux pizerocam3 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux`

`Raspbian Buster (10)`

`RPi Cam Control v6.6.13`

viewed in Google Chrome
`Version 119.0.6045.159 (Official Build) (64-bit)`
on Kubuntu 23.10
`Linux betlogbeast 6.5.0-13-generic #13-Ubuntu SMP PREEMPT_DYNAMIC Fri Nov  3 12:16:05 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

[image]

[image]