silvanmelchior/RPi_Cam_Web_Interface #674: download videos

julianzorro99, 2022-08-04:
Good day
I am making a video with this software, the video lasted about 30 minutes and when I downloaded the video it was 21 minutes long, this is because the downloaded video is going faster than normal, how can I fix this error? 
I appreciate your prompt reply

roberttidey, 2022-08-04:
There are two normal reasons for this type of behaviour.

The first is if you have the box fps different to the recorded fps. The box fps determines the playback rate.

The second is if frames are being dropped. E.g. if you are set to 25fps but frames are dropped so that when played back at 25 fps then it goes quicker. You can check for frame drop by including %k in the annotation which add the frame number to the video.

You should not normally get dropped frames if recording is to the media folder on the SD card unless the card is very slow. You can get frame drop if you have moved the media folder by linking to an external drive or network location. To avoid this you should use the boxing_path support that allows the main recording to be local before boxing asynchronusly to the final destination. See wiki for details.

julianzorro99, 2022-08-04:
Good day
I have not made any changes to the software, nor the path where the videos are saved, just install the software and test it
You could tell me what change I have to make specifically, currently I have a raspberry pi 4b, 8gm of RAM.
Thank you very much

roberttidey, 2022-08-05:
First check that under camera settings- Resolution that the two Video fps (recording and boxing) are set to the same value. They should default to 25

Second, under camera settings - Annotation add a {SPACE}%k at the end of the string and press OK. The live preview should then show the frame number going fast from 0 to 24. Make a short recording and then step through it in an external player. You should see the frame number go from 0-24 repeatedly with no gaps. If you don't know how to check it then post a few second recording here (zipped) and I'll check it.