silvanmelchior/RPi_Cam_Web_Interface #675: The "record video" button does not work

julianzorro99, 2022-08-12:
Good morning.

I have a little problem with the software,
currently I have an Rpi 4, I made a few small modifications to the software and they worked well, I need to lengthen the signal distance then buy a USB Wi-Fi beforehand and connect it by means of the wlan1, and disable the wlan0 forever use the wlan1. But when making this change the "record video" buttons do not work in the right way, not that it may be failing in the software, I greatly appreciate a help.
Thank you very much

roberttidey, 2022-08-12:
Did record button work before changes?

Can you give me more details of the software changes you made?

If understand your wifi changes then you have disabled the on-board wifi and want to use a USB to wifi adapter to get greater wifi coverage is that right.

The log file which you can view under schedule will help tell what is happening when you press buttons.

You should see Capturing started' when you press the record button.

You can also use browser debug (F12) to help see what is going on when you press buttons.

You should see something like
http://192.168.0.23/cmd_pipe.php?cmd=ca%201

when you press the record video button. This is the browser sending a command ca 1 which starts the video recording.

julianzorro99, 2022-08-12:
The recording button works but in a very curious way, if I press the recording button starts recording but the button is not changed to orange to (stop recording) but if I update with f5 the button changes text (not color, only text) and says (stop recording) if I press the button, touch refresh with f5 to change the button again.
The only changes I have made are to rename the buttons and delete some options (attached photo)
Before making the changes worked in an optimal way, when you make the changes were also started well, when you install the wifi antenna with the wlan1 and disable the wlan0, it will generate that problem
[image]

roberttidey, 2022-08-12:
If you turn on browser developer options then you should see a regular call like
http://192.168.0.23/status_mjpeg.php?last=ready

This is what tells the browser what state the server software is in and it has the effect of updating all the button states.

If you have just deleted buttons in index.php without making the corresponding changes in script.js then the script will try to update the status of buttons that no longer exist and this will cause javascript errors that are preventing the script functions completing properly. You will see javascript errors in the browser developer screen console.

You can either change the script to remove any lines that reference non-existent buttons. Or an easier way is just to hie the buttons in index.php rather than delete them. This way the script will continue to work normally but you won't see the buttons you don't want. That is what the index.php does to support the different styles of operation.