silvanmelchior/RPi_Cam_Web_Interface #676: Seeing Page full of texts instead of camera stream

Priya4120, 2022-08-13:
I am able to install and run the module. however, when I reach the webpage I see a page for of text instead of the camera stream. I checked with the raspistill that my camera is working
[image]
.

roberttidey, 2022-08-14:
This is probably an issue with browser set up. For example if javascript has been disabled.

What browser are you using? Have you tied accessing from elsewhere like from a PC or phone?

For most browsers you can turn on developer options (F12) which will show what is going on.

Priya4120, 2022-08-14:
Thanks for the response. I tried accessing the developer options using F12. I am using chrome and Java script is enabled. Here is my output of the console and issues within developer options. Any help is appreciated. 

[image]
[image]

roberttidey, 2022-08-14:
Have you got chrome speechify extension?

If so try disabling this.

Priya4120, 2022-08-14:
That's right. I removed speechify extension. Issues are cleared now but I am still seeing some console errors. Sorry, I am not much experienced on web side. 

[image]

roberttidey, 2022-08-14:
I'll look further into those. I don't get any console errors.

There are references to port 82 and port 291 in your errors> Have you changed port numbers?

init_pt which is referenced doesn't exist in my code.

Elite230, 2022-11-20:
Hi, I am also getting a similar error on chrome without extensions...

roberttidey, 2022-11-20:
I don't see an issue using latest software and using chrome as browser.

Check by turning on developer tools (F12) and looking for errors in console.