silvanmelchior/RPi_Cam_Web_Interface #678: Scheduler doesn't apply to different days settings

simmoinard, 2022-11-30:
Hi, 
I have a problem and I'm not sure if it is a bug or it is a bad settings... ;)

I want to activate motion during specified times, and video capture in case of detection. I want this during the night (from 06:30PM to 08:00AM the next morning) and during the whole weekend

I did this :
[image]

First line to 'disable' the motion and capture on Monday-Friday days
Second line to enable it Sunday & Saturday in Days
Third line to enable everytime after 06:30PM
Fourth line because it looks like after 00:00, the green case (top left in the picture above) is lost. So I "forced it"...

With this it looks like it doesn't work on weekends, it continues just like a week day !
Any idea ? :)

roberttidey, 2022-11-30:
Time periods must be in increasing order for them to work.

Also they should be unique, so the second 8:30 will be ignored. Make it 8:31

simmoinard, 2022-11-30:
Thank you so much !
I did this, can't wait to see this weekend !!

[image]