silvanmelchior/RPi_Cam_Web_Interface #679: Scheduler: the 'Greyworld' value is not accepted for 'wb' White Balance

NicolaP8, 2022-12-23:
Hi,
trying to use the 'wb greyworld' results in an error in the log and the white balance doesn't change.

Version 6.6.26

> [2022/12/23 17:28:51] New period detected 4
> [2022/12/23 17:28:51] Send wb Greymode
> {2022/12/23 17:28:51} Change: white_balance = Greymode
> {2022/12/23 17:28:51} Error: Invalid white balance

roberttidey, 2022-12-23:
Log says you sent wb Greymode.

You should be sending Greyworld

NicolaP8, 2022-12-23:
Ops, you're right. I've done many test and pasted here that with a typo, sorry.

It doesn't accept 'wb Greyworld' as spelled in the Camera Settings page.
But it accepts 'wb greyworld' in lowercase.

roberttidey, 2022-12-23:
Glad that is sorted. Yes those should be lower case; sorry for confusion The web interface has name and value separately so name is Greyworld but value is greyworld