silvanmelchior/RPi_Cam_Web_Interface #680: Pi 4 - force HDMI preview to both monitors

ffcruz85, 2023-01-04:
Is there a way to force the HDMI Preview to go out to both monitors on Raspberry Pi 4?
Both monitors are active at boot (I have hdmi_allow_hotplug:0=1 and hdmi_allow_hotplug:1=1 set), but the preview only shows on HDMI 0.

Thanks for all your work!

ffcruz85, 2023-01-04:
I see on [this commit](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/546bfe853d3cd542a3a0a767d351eedf887d0c87#diff-e8743cabdefbc4ab625748453cb6224ca24143c50e27d8c43d66b336c92ea119) that enabled HDMI preview that there is a line for hdmi_preview 0 in etc/raspimjpeg/raspimjpeg.1  --- Is it a matter of adding HDMI_preview 1 to have it go to second monitor?

roberttidey, 2023-01-04:
I think hdmi_preview just disables/enables hdmi_preview (0 0ff 1 on).

Normally the camera stream goes to a mjpeg stream generator for the web based preview. If hdmi_preview is 1 then a software splitter is used to also send the camera stream to the hdmi port.

I am not sure at the moment about the second monitor as this was done prior to dual monitor support. I'll have a look.

ffcruz85, 2023-01-05:
thanks @roberttidey - yep, I edited my comment after I saw that, my mistake, a little too eager.  I am looking into the docs/code over there as well to see if I can get something out of it, but I appreciate your much more experienced eyes on it. I know this might be a little niche (I am yet again using rpi_cam_web_interface to run an interactive art installation), but I figured that since the option to pick output is there on the raspicam side, there might be some possibility. Thanks again!

roberttidey, 2023-01-10:
Not sure whether this will work but worth a try. It needs both monitors to be the same resolution.

If you go to applications/preferences/screen configuration
Now position the screens on top of each other then apply the change. Reboot. 

That is supposed to duplicate the screens.