silvanmelchior/RPi_Cam_Web_Interface #681: Is Camera Module 3 compatible?

canuckdev, 2023-01-23:
Is [Camera Module 3](https://www.raspberrypi.com/products/camera-module-3/) compatible?

roberttidey, 2023-01-23:
No. I don't believe so. Module 3 requires using the libCamera interface and not the original MMAL interface used by this software.

icooper9000, 2023-02-03:
Thats a pity, module 3 camera is so much better. Are there any plans to support module 3 and older cameras as well in the future, perhaps by detecting the camera fitted during nstallation and then loading the appropriate interface?

kaihokori, 2024-10-26:
Hi @canuckdev, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). Testing was conducted on the camera module 2 and Arducam OwnSight 64 MP camera, but given that this approach uses Picamera2 (developed by Raspberry Pi themselves), it should be compatible with the module 3. Be aware that not all functionality has been implemented but it should be a good starting point ^_^