silvanmelchior/RPi_Cam_Web_Interface #683: SimpleOn and SimpleOff button not documented

hogwell, 2023-03-22:
On the System tab on the main RPICam Web interface there is a button labeled SimpleOn or SimpleOff.

I cannot find this documented on the wiki or anywhere and it appears to just toggle between states on the button without any effect.

What is this button meant to do?

PCAssistSoftware, 2023-03-22:
turning it on shows a Simple button in top right hand corner to allow you to switch to screen just showing camera and no menus, and then can click Full button to switch back 

[image]

hogwell, 2023-05-11:
Thank you.
Is this documented on the wiki somewhere?

roberttidey, 2023-05-11:
It is described in the section 2.1 Basic usage

hogwell, 2023-05-12:
Ok, I see it now... I was searching the wiki for 'SimpleOn' (the button label).
Thanks.