silvanmelchior/RPi_Cam_Web_Interface #685: Password

iggymorr, 2023-04-24:
I put I password on it but could access the video or anything it didn’t even show a place to put my user and password on the site not sure if I did something wrong please help

roberttidey, 2023-04-24:
First restore normal operation by running the install.sh script again setting user to a blank field.

Which Web server are you using? Apache is the one I do most testing on. It is possible that if you are using a different one then something has changed in how that handles authorisation.

iggymorr, 2023-04-25:
i redid it still not showing up to use a password

roberttidey, 2023-04-25:
I have just done a test on a clean install of Bullseye I first ran install.sh with no user/password, tested OK. Then I reran install.sh with a user/password.. I restarted my browser to make sure it was starting clean then accessed the camera and got a sign in form that needed me to enter user/password to get normal access. That worked fine.

Are you getting any errors when you run the install.sh script? There should be some lines in there saying adding password for user and then updating password.

iggymorr, 2023-04-26:
I removed the cam sub folder that worked I think the sub folder was not working i got it to work :)