silvanmelchior/RPi_Cam_Web_Interface #688: Project Status

wallarug, 2023-11-03:
Hi there,

I just wanted to check in on the status of this project?   

I have noticed there has been no updates in 12 months.   Does someone need to pick up this project to keep it going?

I really really love this project and have used it whenever I fire up a Pi Cam.  Would be sad to see it decay away.

roberttidey, 2023-11-03:
As the sole developer I still use the software extensively myself and will investigate issues that are raised.

The main issue is that it currently relies on the camera support built into the GPU via the MMAL interface. This is no longer directly supported by the Raspberry OS although it can be turned on by turning on the legacy camera support. Unfortunately this also means the range of cameras supported is limited.

To switch to using the libCamera method used in the OS for all cameras now is a fairly significant undertaking needing quite a lot of effort on coding of the main raspimjpeg process which supports the camera interface and all the video processing. There are also some technical challenges around the motion detection as that uses the h264 vectors available in the MMAL interface. That gives very efficient motion detection and there is no direct equivalent in libCamera.

Currently, this not a development I can undertake on my own. If there are other developers who are interested then I would be very willing to contribute to this effort.

wallarug, 2023-11-03:
Hello @roberttidey ,

Good to hear from you so quickly!   I'm happy to help out a little bit.   

I did understand that there was a significant rework required since the OS changed in c. 2021.  I probably need to have a more in-depth look at the inner workings before asking questions, but I'll ask anyway.

1. Is there an alternative to using `raspimjpeg` to get the stream out of the camera and into the web process?  
2. Is there a way to use the RPI supported methods to get the footage out of the camera (libCamera) and into the web?
3. Is the core problem that the `libCamera` suite is not GPU optimised (hence using alternatives)?

I'm trying to understand where someone would start.  There seems to be lots happening here that needs attention.

I have not found any alterative to RPi_Cam_Web_Interface that is nearly as feature rich or easy to deploy.  It would be great to help fix it up and see it restored to it's former glory.

roberttidey, 2023-11-03:
raspimjpeg (source at https://github.com/roberttidey/userland/tree/master/host_applications/linux/apps/raspicam, files starting with RaspiM) was originally based around the original raspberry camera support raspivid. It was compiled within the userland repository as that provided all the necessary MMAL references.

1 & 2) raspimjpeg exploits the pipelining process in MMAL to get several streams out of the camera simultaneously and process them in different ways. One provides the resized mjpeg stream to feed the main preview. Another stream is used for full resolution video recording to h264 /mp4. A third stream allows for still image capture. A fourth stream provides h264 motion vectors for use in the motion detection. As well as the main camera / video processing. raspimjpeg supports a number of commands fed into command pipelines. The web interface uses these commands to provide the overall functionality.

So the approach I would take is to modify the camera / processing in RaspiMCam.c and the motion detection in RaspiMMotion.c while leaving all the main control / command fairly intact. The source code would then be compiled separate from userland but linking with libCamera instead.

The camera settings that the web interface exposes would also need to change as these are specific to those defined in MMAL and would need to change to those that libCamera uses. The web side would need to change to reflect that but that part would be fairly straightforward.

3) libCamera does use GPU for some of the intensive tasks so I don't think that is a fundamental problem. I suspect porting to this environment would be more cpu intensive as I think the pipelining / stream splitting will involve more of the cpu in data movement. This was originally developed to run well even on Pi Zero v1 and it still does with quite low cpu usage. I also wanted it to very efficient so the pi could do other stuff. Later models including Zero v2 are much more powerful so this becomes less of a problem. The one area that would need thinking about is the motion detection. Originally an external "motion" process was used for this which worked by comparing frame buffers for pixel differences. This works OK but can use quite a bit of cpu. Getting the motion vectors used as part of the h264 encoding allowed the use of the internal motion detection method which is more effective and very efficient.

Getting into this requires understanding the RaspiM files. In particular, RaspiMCam.c which would need almost a complete rewrite.

Even with a new libCamera version, I think I would archive the original raspimjpeg based version using MMAL particularly for any Zero v1 uses if the new version strained that platform too much.

wallarug, 2023-11-23:
Do you know if @silvanmelchior is still around to help on this project?

silvanmelchior, 2023-11-24:
Hi there, still around, but unfortunately no capacity here

SamVimes78, 2024-01-23:
Just want to let you guys know that I'm really excited to read there's still some hope for this project to support libcamera maybe sometime in the future.

I bought a camera module v3 because i thought i could just replace my v2 module and keep using the RPI-Cam web interface. Was really sad when I found out it doesn't just work that way...

Wouldn't even need motion detection. Just a simple web interface to manually record videos and take photos with the v3 module would be so much appreciated!

wallarug, 2024-01-23:
> Just want to let you guys know that I'm really excited to read there's still some hope for this project to support libcamera maybe sometime in the future.
>
> I bought a camera module v3 because i [...]

Hey @SamVimes78.  Great to see interest.  

I've actually put this in as a research project for Semester 1 with some University Students to help kick start progress.  Hopefully they are up for the challenge to help bring this project back!

wallarug, 2024-08-06:
Hey @roberttidey  I now officially have two teams of students to help with this.  Are you ok to provide some background on the project and potential new approaches to help bring this to the new hardware at the right time?

I am very appreciative of your help to date.

roberttidey, 2024-08-06:
Yes. I will try to assist in any way I can.

It would be good to find a way of replacing the legacy camera support with libcamera method as that would open it up to a wide range of camera hardware and be properly supported on later versions of the the Raspberry OS including 64 bit versions.

The main efforts would be focussed around providing a rough equivalent of the raspimjpeg basic module. It doesn't have to be directly equivalent as any changes on the web interface tend to be fairly straightforward.

wallarug, 2024-08-06:
> Yes. I will try to assist in any way I can.
>
> It would be good to find a way of replacing the legacy camera support with libcamera method as that would open it up to a wide range of camera hardware [...]

I tend to agree.  I'm not interested in rebuilding the UI.  Just the 'behind the scenes'.   

I'll get in touch with you early next week for a meeting / call with them.

moataz-youssef, 2024-08-07:
> Hey @roberttidey I now officially have two teams of students (12 people) to help with this. Are you ok to provide some background on the project and potential new approaches to help bring this to the new hardware at the right time?
> 
> I am very appreciative of your help to date.

This is an awesome development. Thank you @wallarug for this and thanks to @roberttidey and @silvanmelchior for the original work.

wallarug, 2024-08-10:
> Yes. I will try to assist in any way I can.
>
> It would be good to find a way of replacing the legacy camera support with libcamera method as that would open it up to a wide range of camera hardware [...]

Hey Robert.  I have the 3 teams ready to go.  Are you free sometime this week to do a quick presentation on some topics?

- Existing software
- Why the software is broken / incompatible
- Potential new approaches to be explored
- Your best approach / bet on being successful.

How do I get in touch with you directly?  Can you send me an email?  wallarug@coliemore.com.au

wallarug, 2024-08-11:
Hey @roberttidey  - I've sent an email response to you.  Might have gone to junk.

wallarug, 2024-09-10:
@roberttidey @moataz-youssef @SamVimes78 

An update:  https://youtu.be/sXU2rHyhO38?si=wVajOVvG8J0EGDJ4&t=353

roberttidey, 2024-09-10:
Thanks for sharing the update video.

It looks like a lot of good progress has been made and the approach certainly brings it up to date.

kaihokori, 2024-10-25:
Replying to @roberttidey @moataz-youssef and @SamVimes78

Just as some background, @wallarug tasked three university teams to take different approaches to updating the raspimjpeg system. Above this comment you'll find the contributions @dylan-lom and his team made. However, the progress you've previously been filled in on came from my team, so we'd also like to provide a final update on @wallarug's behalf: 

https://www.youtube.com/watch?v=nE2B7n2fVq8

TLDR for the video; Our approach was to use Python with Picamera2, Libcamera and OpenCV. A significant amount of previously existing features has been implemented along with some new ones too. 

You can find our approach and instructions in [our repository](https://github.com/Windermere-Technology/RasPyCam).

wallarug, 2024-10-27:
Thank you @kaihokori for your note.

As mentioned - There were three teams working on brining the `RPi_Cam_Web_Interface` back to life.

I have reviewed all the work and can confidently say that there are now multiple options for continuing this project.

As a bonus, the software was tested on both RPI 5 and RPi Zero 2.
They also tested RPi Camera v2 and an @arducam OwlSight 64MP Sensor to prove that future cameras supported by `libcamera` will be supported in this package as well.

For those who have been following this saga, you can watch and review the 3 demos here.

Team A - https://www.youtube.com/watch?v=nE2B7n2fVq8
Team B - https://www.youtube.com/watch?v=Kh2dnlxGJGI
Team C - https://www.youtube.com/watch?app=desktop&v=eohqJpkeJbk

The next steps is to integrate these into RPi_Cam_Web_Interface and launch a new release.

I'm pretty flat out at the moment - so any community help to merge these together would be much appreciated.

Thank you @roberttidey  for providing advice in the early stages of the project.

roberttidey, 2024-10-27:
Wow. Good to see all the progress on this.

I look forward to playing around with this in the near future.

R-Rosenow, 2025-02-28:
Any news? I would also like to operate an Arducam with the program.

roberttidey, 2025-02-28:
As reported earlier in this thread their has been a significant development effort in Australia to allow similar functionality and user interface to be achieved via the libCamera interface which allows the use of a wide range of cameras.

Check out

https://github.com/Windermere-Technology/RasPyCam

R-Rosenow, 2025-02-28:
> As reported earlier in this thread their has been a significant development effort in Australia to allow similar functionality and user interface to be achieved via the libCamera interface which allows the use of a wide range of cameras.
> 
> Check out
> 
> https://github.com/Windermere-Technology/RasPyCam

This isn't new to me. I also installed RaspyCam, but it still doesn't work.

wallarug, 2025-03-01:
> This isn't new to me. I also installed RasPyCam, but it still doesn't work.

Hey @R-Rosenow  - what aspect doesn't work?

I've two alternative methods that I can release later today.  I thought I already did it, but must have missed it.

R-Rosenow, 2025-03-01:
> Hey [@R-Rosenow](https://github.com/R-Rosenow) - what aspect doesn't work?
>
> I've two alternative methods that I can release later today. I thought I already did it, but must have missed it.

RasPyCam does not recognize the camera. I can take pictures and videos using the libcamera command in the terminal. I've read the instructions from RasPyCam on Github, and the description is a bit sparse.