silvanmelchior/RPi_Cam_Web_Interface #689: Error when installing

05k4rausdertonne, 2023-12-01:
I'm trying to install on a raspberry pi zero 2 w running raspbian bookworm and i get this error:

```
Package gpac is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gpac' has no installation candidate
```

i looked everywhere online but havent found a way to install gpac.
and yes i ran `apt upgrade` etc multiple times :D
i'm going to try to build it myself now but this shouldn't be the way to install imho.

I would be happy to hear if this is an issue for others aswell or if i missed something obvious.

regards ;)

emjas92, 2023-12-01:
I came across this recently too, and went down a rabbit hole trying to resolve it.
Basically the gpac packages have not yet been made available in Raspberry Pi Bookworm.

The only solution for now is to use Bullseye until Raspberry Pi foundation adds the packages.
I'll try find the reference for this,

mlambi, 2023-12-05:
I have read from several people that Bookworm is not ready for prime-time.
Lot’s of problems to work out.

Bra1nsen, 2024-02-17:
any updates here, same issue: 

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Package gpac is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package php7.0-cli
E: Couldn't find any package by glob 'php7.0-cli'
E: Couldn't find any package by regex 'php7.0-cli'
E: Unable to locate package libapache2-mod-php7.0
E: Couldn't find any package by glob 'libapache2-mod-php7.0'
E: Couldn't find any package by regex 'libapache2-mod-php7.0'
E: Package 'gpac' has no installation candidate
```

I come from: 

https://www.remote.it/resources/10-minute-secure-webcam-with-raspberrypi-and-remote-it

Kind regards

roberttidey, 2024-02-17:
First this is probably because you have a later version of php. If you have used bookworm for example php version 8.2 is installed by default. You can check by running php -v  in the command window.

You can specify the php version when the install script runs.

The install script tries to figure out which php version to run but hasn't been updated for bookworm. I'll check that out.

gaozihan4, 2024-03-04:
I've tried to rebuild from this 
https://github.com/gpac/gpac/wiki/GPAC-Build-Guide-for-Linux

Also, remove the 'gpac' from shell command 'apt-get' in 'install.sh', which I believe actually caused this issue.

Magnethead794, 2024-06-25:
I am having this same issue. Apache is installed and running on port 80; Node-Red is installed and running on port 1880; PHP 8.2 is installed and running. But I still get "gpac has no installation candidate". I've ran updates and such.

Bra1nsen, 2024-08-11:
When will this issue be solved? Whole RPi Webcam isnt usable, which is sad.

05k4rausdertonne, 2024-08-11:
just use legacy OS (RPi OS "Bullseye" i think). thats how i got it running. i don't know if this is gonna be fixed soon.
we shouldn't be so demanding of open source devs. most of them do this in their free time after all.

ebretteville, 2024-08-11:
> just use legacy OS (RPi OS "Bullseye" i think). thats how i got it running. i don't know if this is gonna be fixed soon.
> we shouldn't be so demanding of open source devs. most of them do this in their free time after all.

You're so right

Bra1nsen, 2024-08-11:
I'm not making any demands; I'm simply outlining the current situation. What adjustments are necessary to ensure compatibility with the latest Raspberry Pi OS (Bookworm)?

ebretteville, 2024-08-11:
https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688

kaihokori, 2024-10-26:
Hi @05k4rausdertonne and others, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). Be aware that not all functionality has been implemented but it should be a good starting point ^_^