silvanmelchior/RPi_Cam_Web_Interface #692: Compatibility with raspberry pi camera module 3?

biprav, 2024-02-10:
Hi.
I just bought a 12 MP raspberry pi camera module 3 and have connected it with a Pi zero 2W. I have tried using the cam web interface with the camera module 3 but it gives various errors. Can you kindly tell me whether it works with camera module 3?

Thank you.

roberttidey, 2024-02-11:
This software uses the GPU legacy camera support which only supports v1 and v2 module rapsberry cameras.

biprav, 2024-02-11:
> This software uses the GPU legacy camera support which only supports v1 and v2 module rapsberry cameras.

Okay. Thank you for responding.

ways, 2024-05-23:
No workaround to pipe video from camera v3 to this?

kaihokori, 2024-10-26:
Hi @biprav, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). This program should be usable on the pi camera module 3 as it uses Picamera2 (which is developed by Raspberry Pi themselves) as a foundation. Be aware that not all functionality has been implemented but it should be a good starting point ^_^