silvanmelchior/RPi_Cam_Web_Interface #694: libmmal not found issue.

rewolff, 2024-03-11:
I followed the instructions but skipped enabling the camera as I had already verified the camera working. 

Then... "no images". Some debugging shows: 
```
tanky:~/RPi_Cam_Web_Interface> sudo su - www-data
www-data@tanky:~$ raspimjpeg
raspimjpeg: error while loading shared libraries: libmmal_core.so: cannot open shared object file: No such file or directory
www-data@tanky:~$ 
```
A google search for this error shows that I need to enable legacy support for the camera. Then the install document says to enable that in "raspi-config". So I ran raspi-config and found that "interface options" is no longer option 5 and the camera is not mentioned there. 

I figured that one out before: Modern kernels auto-detect the camera, so it is enabled automatically, no longer requires a raspi-config entry. 

So I downloaded a recent OS, should I have chosen an older one? There are plenty of hints that bookworm is supposed to be working.

roberttidey, 2024-03-11:
Bullseye is OK for this software.

I don't know about Bookworm. They were threatening removing the legacy camera support at some time.

rewolff, 2024-03-11:
So "bullseye" is older, I indeed have to downgrade? 

I have spent quite some time on getting everything ready. The camera stuff is not quite "the only thing" going on on this raspberry. :-(

roberttidey, 2024-03-11:
Raspberry OS now have moved to a different way of interfacing to cameras (libCamera) instead of using the MMAL interface which had been used for all Raspberrys since they were introduced.

This software relies on the MMAL interface which is supported via the legacy camera option up to Bullseye but is not present in bookworm.

It is a lot of work to change this software to use the libCam interface, particularly as the MMAL interface provided some facilities which are not present in libCam.