silvanmelchior/RPi_Cam_Web_Interface #695: No support for Raspberry Pi Global Shutter Camera?

boos87, 2024-04-24:
I have a Raspberry Pi Zero 2 with a raspberry pi global shutter camera, but the camera doesn´t work wir the RPi Cam Web Interface. 

A Error Message appears, if i open the web interface. 

Error in RaspiMJPEG
Restart RaspiMJPEG (./RPi_Cam_Web_Interface_Installer.sh start) or the whole RPi.

Can somebody help me, please???

roberttidey, 2024-04-25:
This software only supports cameras via the MMAL interface (now called legacy support). This includes Pi camera v1 and v2.

The Global shutter camera is not supported.

Bobo-amg, 2024-05-16:
I think you should publish this info on your website. Not everyone looks at github et al.

kaihokori, 2024-10-26:
Hi @boos87, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). This approach does include a setting for adjusting the camera shutter speed. Be aware that not all functionality has been implemented but it should be a good starting point ^_^